lgsvl/simulator #1598: How to run API Only sim with a custom vehicle and Apollo modular testing?

krtt, 2021-07-16:
I have prepared a test scenario script that is able to run ok with our custom map and SVL default Lincoln vehicle, using the UUID of the vehicle's Apollo 6 modular testing (2e9095fa-c9b9-4f3f-8d7d-65fa2bb03921). Now, I would like to run the same scenario with our custom vehicle and modular testing only, but the car does not start driving. In some point the dreamview reports also a warning that the Prediction module is not running. But when I run it with Lincoln car, then no such message appears. I compared the car configurations but don't seem to notice anything that needs changing. What am I missing?
The script and both car configurations attached.
[test_scen_script_and_vehicles_conf.zip](https://github.com/lgsvl/simulator/files/6831888/test_scen_script_and_vehicles_conf.zip)

EricBoiseLGSVL, 2021-07-16:
@lemketron ? any idea?

lokesku, 2021-07-25:
Apollo_UT_Lexus.json in attached zip file is not a modular testing configuration. For modular testing configuration, please make sure you include [3D Ground Truth Sensor](https://wise.svlsimulator.com/plugins/profile/e3fef197-9724-48ec-b98e-c5a0892d09c4) and [Signal Sensor](https://wise.svlsimulator.com/plugins/profile/c4ba9b81-b274-4c05-b78e-636e61b4590e)

Steps to do this with custom vehicle:
1) Upload custom vehicle to cloud
2) Create sensor conf on custom vehicle: copy json from [Apollo 6 modular testing sensor configuration](https://wise.svlsimulator.com/vehicles/profile/73805704-1e46-4eb6-b5f9-ec2244d5951e/edit/configuration/2e9095fa-c9b9-4f3f-8d7d-65fa2bb03921)
3) Disable Perception & Traffic Light modules of apollo on dreamview and enable others

For more details on modular testing sensor configuration and run follow: https://www.svlsimulator.com/docs/tutorials/modular-testing/

lokesku, 2021-07-25:
If you need further help, it will be helpful to be able to reproduce your scenario. To do this, we would need your simulation configuration, map, vehicle and sensor configuration. You can simply use share button on your simulation on https://wise.svlsimulator.com with contact@svlsimulator.com

krtt, 2021-08-09:
We got it working. Sorry for late reply. It's ok to close this.

closed 2021-08-09 by krtt