lgsvl/simulator #1599: How to make a map from point cloud data?

nparkstar, 2021-07-19:
I am trying to make new a map from point cloud data. But I can't find detail process. What is the right process?

My guess : 
1. Create a new scene
2. Import point cloud
3. Annotation
4. Build Environment
5. Upload to cloud
6. Simulation

I saw the document and video(https://www.svlsimulator.com/docs/simulation-content/add-new-map/). That starts from importing assets from Unity store. I want start with only point cloud file.
What is the way?

EricBoiseLGSVL, 2021-07-23:
pretty close.
2) import point cloud, be sure to generate mesh to be able to annotate.
Other than that the list is correct.

nparkstar, 2021-07-23:
> pretty close.
> 2) import point cloud, be sure to generate mesh to be able to annotate.
> Other than that the list is correct.

Thank you for your answer.
And I did it.
I came to know I can build environment without any annotation.

closed 2021-07-23 by nparkstar

EricBoiseLGSVL, 2021-07-23:
yep, you just need the annotation holder objects but they can be empty. Glad you got it working

nparkstar, 2021-08-31:
> Hi,
>  <https://user-images.githubusercontent.com/82795532/130917492-89074b2c-ab3d-4450-9351-3240d0c8bf0e.png>
>
> I used the MapToolBox to import the point cloud（.pcd）.
> Could you please tell me the steps to use this map in the simulator?
> Setting a mesh collider on this map will result in an error.

Hi, 

I think you try to upload *.pcd file to SVL Cloud.

You should make Environment asset in Unity Builder first and then can upload asset file to SVL Cloud.

 

Good luck,

EricBoiseLGSVL, 2021-08-31:
What is this comment? Is the issue not resolved? @nparkstar

nparkstar, 2021-09-01:
I’m sorry, I made a mistake.

I received a mail as below.

 

I think I replied to who send to me.

 

The sender may send me a mail because the issue was closed.

 

Is any way to reply to the sender?

 

I’m sorry for bothering you.

 

 

 

The mail I received.

 

Hi,
 <https://user-images.githubusercontent.com/82795532/130917492-89074b2c-ab3d-4450-9351-3240d0c8bf0e.png> 

I used the MapToolBox to import the point cloud（.pcd）.
Could you please tell me the steps to use this map in the simulator?
Setting a mesh collider on this map will result in an error.

EricBoiseLGSVL, 2021-09-01:
no worries, just reply in the website.  Email spams the response.