lgsvl/simulator #1600: Trying to run autoware.auto alongside SVL simulator

tecena, 2021-07-19:


EricBoiseLGSVL, 2021-07-19:
@hadiTab or @lemketron ?

lemketron, 2021-07-19:
@cyberphysicalrobotics Any thoughts on this?

mslavescu, 2021-07-24:
Try my demo script linked in video description, it will start SVL and Rviz2 as show in the video:
https://www.youtube.com/watch?v=Bkk4OYjdnSU

Make sure you update ADE to latest version, by first running this command (one time only):
ade --rc .aderc-lgsvl start --update --enter

marcusvinicius178, 2021-08-03:
Hi @tecena the issue you are facing is due to the wrong documentation: https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/installation-ade.html

Probably you have followed the following step:

`git checkout tags/1.0.0 -b release-1.0.0`

You should not checkout for this branch because it is missing the **autoware_auto_msgs** package (the error you are facing), therefore your AutowareAuto workspace will never be built!!! **Unless ** you dowload this repo manually in the link https://gitlab.com/autowarefoundation/autoware.auto/autoware_auto_msgs/-/tree/master/autoware_auto_msgs and paste it inside the AutowareAuto/src/external folder (where it should be when you clone the repository, but for some reason the AutowareAuto team forgot or removed this folder in the 1.0.0 branch). Even if you put manually this package you will keep facing missing packages issues, as I also had. The next missing folder is OpenCv. And you will have this error: 

[image]

This folder you will also need to add manually (dowload, copy and paste) or you can get it cloning this repo:
https://github.com/opencv/opencv

you should:

```
cd adehome/AutowareAuto/src
git clone https://github.com/opencv/opencv.git
```

Now you must modify the **.aderc** file inside the /adehome/AutowareAuto folder
`gedit .aderc`
and paste this content below to get the last simulator release:

```
export ADE_DOCKER_RUN_ARGS="--cap-add=SYS_PTRACE --net=host -e RMW_IMPLEMENTATION=rmw_cyclonedds_cpp"
export ADE_GITLAB=gitlab.com
export ADE_REGISTRY=registry.gitlab.com
export ADE_DISABLE_NVIDIA_DOCKER=true
export ADE_IMAGES="
  registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto/amd64/ade-foxy:master
  registry.gitlab.com/autowarefoundation/autoware.auto/autowareauto/amd64/binary-foxy:master
  registry.gitlab.com/apexai/ade-lgsvl:latest  
"

```

Afterwards enter again the docker image:
`ade start`
`ade enter`
`cd AutowareAuto`

The main step now is **checkout again to the master branch** to "escape" the branch 1.0.0 (or just remove all docker images, and restart installation without checkout to this branch) !

`git checkout master`

you will face a lot of errors (probably), that you should add, remove, merge or stasch the different files not compatible between the branch and the master. So go ahead and do what you prefer. I have added the different files, before commit the new branch:

`git add (TAB to see what is missing)` then press Enter one by one missed file
`git stash save release-1.0.0` I have named "release-1.0.0" this command will hide this branch and you can finally checkout again to the master branch  (where the correct packages are).
`git checkout master`
Then build again the workspace
`colcon build`
You still have issues in the compilation (I had at least again with OpenCvConfig.Cmake). Now you **finally** run the commands below:

`rm -rf build/ install/log/ src/external/`
`vcs import < autoware.auto.$ROS_DISTRO.repos`

I believe the last command **finally get the missed packages** that raise the error you faced! The the last command, build again!!
`colcon build`

And you will finally build the 119 packages from AutowareAuto.

I still did not test if it is going to work with the new SVL release, I am still trying.