lgsvl/simulator #1602: Simulator container creation issues

shivpip131, 2021-07-20:
Hi Team,
If I am running docker file of simulator and building image, after that I had run the container , now the container is running .
How can I verify that my  container is having running  #simulator?

EricBoiseLGSVL, 2021-07-23:
We are working on better docker support and should release soon with a console in simulator to see docker output. @eagafonovlge might have more helpful tips for docker now.

shivpip131, 2021-07-23:
Hi , team 
I have taken  docker file from GitHub/simulator/2021.1 and I have a azure VM , in this VM I have docker installed, nividia-smi installed,nividia tool kit installed, now I am  building docker svlsimulator image by the same command which has given in your GitHub readme file , after that I had run the container , now the container is running .
Questions :- How can I verify that my container is having running #simulator?
I have verified by localhost:8080 in the browser but it doesn't work.

Please help me out team.

shivpip131, 2021-07-26:
Hi lg team,
I am running simulator as a container by the below command 

$ docker run -ti \
     --gpus=all \
     --net=host \
     -e DISPLAY \
     -e XAUTHORITY=/tmp/.Xauthority \
     -v ${XAUTHORITY}:/tmp/.Xauthority \
     -v /tmp/.X11-unix:/tmp/.X11-unix \
     -v ~/.config/unity3d:/root/.config/unity3d \
     svlsimulator[:<VERSION>]

Now when I am access the container by the command
Docker exec -it containerid  /bin/bash 
And running inside container   ./simulator 
Getting  the below error:- 
segmentation fault(core dumped)

EricBoiseLGSVL, 2021-07-26:
More than likely you are not using the GPU and it causes a crash.  What GPU are you using?  What is the player.log from the simulator application? More info on what is going on is better than creating duplicate issues. Please do not do this.