lgsvl/simulator #1603: Coredump while start the second simulator instance.

deepalcoholic, 2021-07-20:
After I install the building from source environment. 
Coredump happens while starting the second simulator instance using the released bundle.
Here is the log 
[Player.log](https://github.com/lgsvl/simulator/files/6846138/Player.log)
Would anyone encounter this issue? Need your help, thanks a lot!

EricBoiseLGSVL, 2021-07-23:
Are you altering the data_path ```data_path: C:\Users\USER\AppData\LocalLow\LGElectronics\SVLSimulator02``` in the config.yml file for the second simulator instance?  We don't want users to have multiple simulators overwriting the same cached assets so we require different data paths.

deepalcoholic, 2021-07-27:
@EricBoiseLGSVL Thanks for your reply.
I am using the Ubuntu18.04.  
I have not find the data_path config item in the config.yml, there are only the following items:
headless: false
read_only: false
api_hostname: "localhost"
api_port: 8181
cloud_url: "https://wise.svlsimulator.com"
Should I add the data_path item into the config.yml?
Thanks again.

closed 2021-07-27 by EricBoiseLGSVL