lgsvl/simulator #1604: Simulation on remote computer

arudomanenko, 2021-07-20:
Hi! I have a two computers: powerful one and mine) I configured cluster on the first and now I can to run simulation on it. Its is great. But I want to run simulation on my computer. Where can I find the tutorial about how can I do it? It seems that I should to add cluster from first PC, but I can't do it (I just can't see option like that).
Lgsvl version: 2021.1 (on both computers)
Thanks

arudomanenko, 2021-07-20:
The solution is VERY easy - just log in by you account on server and run lgsvl_bridge in your computer

closed 2021-07-20 by arudomanenko

roshambo919, 2021-10-05:
@arudomanenko does this procedure require a GUI, or can it be accomplished with only terminal access (SSH) to a powerful machine? I'm not able to figure out how to link a remote computer through ssh.