lgsvl/simulator #1609: Simulator starts on the "Offline" mode and does not response to "Go Online"

manone888, 2021-07-20:
Hi guys, when I launched the simulator, it showed the below screen:
[image]
It did not have any response when I clicked "Go Online" and "Unlink". But it had response to the "Clear Cache" and "Quit".
I tried the three releases of 2021 and got the same result. But when I tried the release of 2020.06, I could click the "Open browser" button.
Could you please advise on this issue? Thanks a lot.

EricBoiseLGSVL, 2021-07-22:
what version is the image from 2021.2?  What cloud url are you trying to reach? It should be in the config.yml file in the root.

nsrj, 2021-07-23:
Thank you for your reply @EricBoiseLGSVL. Yes, it is from 2021.2. Please see the below figure.
[image]
By the way, the OS of the computer is Ubuntu 20.04.

manone888, 2021-07-23:
@EricBoiseLGSVL, thank you for your reply. Yes, it is from 2021.2. Please see the below figure.
[image]
By the way, the OS of the computer is Ubuntu 20.04.

EricBoiseLGSVL, 2021-07-23:
Please navigate to the AppData folder for simulator and delete it and start it again.  Another issue maybe something in the OS is blocking the app from the network.

baoxinchen, 2021-08-01:
Hi @manone888, are you using a computer without the cable internet interface? Like, gaming laptops. I had the same issue on my razer laptop. If I plug in a USB internet dongle, the simulator could go online on WiFi or wire network. I think this is a BUG on this Linux build.

EricBoiseLGSVL, 2021-08-02:
Yes, @baoxinchen is correct, direct cable internet is a must.  We have no support for wireless or usb internet hubs.  It would be great if you post a solution for users if they are using this.  Next release we have an update to web socket sharp that may work better.