lgsvl/simulator #1613: SimulatorStatus never Running while offline

RobShewagaUHN, 2021-07-21:
SVL 2021.1

It looks like the current simulation state can be found from Loader.Instance.Status, but it is never set to Running if the simulator is running in Offline mode. Status is set by Loader.reportStatus, but the only time it is asked to set itself to Running is behind an offline lock-out if statement (Loader.cs).

        if (Instance.CurrentSimulation != null && ConnectionManager.Status != ConnectionManager.ConnectionStatus.Offline)
        {
            Instance.reportStatus(SimulatorStatus.Running);
        }

Is there any reason for this? Will anything else in particular break from removing the second half of the if?

Thanks

EricBoiseLGSVL, 2021-07-21:
In 2021.2 we have fixes for this, where in Loader.cs is this? Please post line numbers

RobShewagaUHN, 2021-07-21:
Line 733 in Loader.cs, as well as line 114 in LoadScene.cs

EricBoiseLGSVL, 2021-07-21:
wait to make the version jump when we push a fix this week. Sorry about the delay