lgsvl/simulator #1614: Center Lane Position

MartinaB85, 2021-07-22:
Good afternoon,

is there a way to get the position of the lane center inside the script NPCLaneFollowController.cs/VehicleController.cs ?In which script is this info located?

Thank you very much in advance for your help,

Martina

EricBoiseLGSVL, 2021-07-22:
map lane world positions are the center lane positions.  It also has other data if you need it.  Is that what you are asking? If you are not on a waypoint position, you could calculate with prev and current index.

MartinaB85, 2021-07-23:
Ok, thanks, will look into that if I can achieve what I want.

Thank you.!

MartinaB85, 2021-09-03:
Good Morning,

how can I continuously get the position of the white line, which divides each lane, as the Ego car is following the lane?

Thanks,

Martina

EricBoiseLGSVL, 2021-09-03:
Hello, MapHolder has access to all annotation objects in scene.  You could ask MapManager for closest lane, get the left and right boundary line of this lane and then calculate the position on the line perpendicular to the map lane line.