lgsvl/simulator #1615: Blank Unity Editor when building from source Windows

Jasabl19, 2021-07-22:
**System Information**
[image]

SVL installed via this [guide](https://www.svlsimulator.com/docs/installation-guide/build-instructions/) 

**Description of issue**
I was successfully able to get to the end of step 2. At the beginning of step 3, the unity editor window should look like this: 
[image]

Instead, my unity window looks like this:
[image]

Weirdly enough, if I finish the build steps, the simulator DOES run and operate properly. The issue is that I'm trying to follow[ this guide](https://www.svlsimulator.com/docs/simulation-content/add-new-ego-vehicle/) in order to create a custom ego vehicle. Since my inspector and hierarchy panes are blank, I am unable to properly import the vehicle mesh. Any advice or troubleshooting steps are appreciated!

Thanks,
Jacob

EricBoiseLGSVL, 2021-07-22:
You need to click open the Simulator menu to select the build menu.  Then dock it using drag and drop with the top tab of the window like the image in docs or just leave it floating.

Jasabl19, 2021-07-22:
Thank you for your quick response. This is what my build menu looks like: 
[image]

As I stated - I was able to complete the build and even run the sim. My question is mainly why in the tutorial is the central window showing the splash page and mine just shows an empty world.

Also, I do not seem to have the same hierarchy options that are shown in the tutorial and my environments and vehicles are completely empty - do I need to manually add those in order for anything to come up?

Thanks again for your time - I'm new to this.

lemketron, 2021-07-23:
It looks like some things were done in the Unity environment (arranging of tabs, opening of the Loader scene, etc.) before the screen shot was taken for [step 3](https://www.svlsimulator.com/docs/installation-guide/build-instructions/#build-stand-alone) which says "Open the Simulator menu from the top toolbar and select Build... to open the build window (shown below with a red outline)"

It doesn't actually matter because what was important in that step was opening the Simulator's Build menu in order to build the simulator executable but I can understand why the screen shot might have been confusing since it didn't match what you were seeing.

EricBoiseLGSVL, 2021-07-23:
No worries, it's great progress to be able to create a local simulator binary already :)

You need to clone the repos we provide if you want to build custom maps or create your own assets.  Please see docs on how to do this [here](https://www.svlsimulator.com/docs/simulation-content/add-new-map/).

Once you have correctly created the map/vehicle assets in the editor then you will see them in this drop down to build.  These options are only needed if you want to make custom assets.  You then can upload to the webui to run in your simulations.

The images shown are just a working dev environment in a scene.  The doc images are circling the important aspects.  Don't worry about matching images exactly.  Sometimes they are from earlier versions of simulator and not a perfect match.

I recommend watching a few videos on Unity and how the engine/development works before diving into asset creation.  We try to explain as much as we can but content creation is not easy.  Just open new issues when you get stuck or have a problem.  Good luck!