lgsvl/simulator #1616: bridge connection between Apollo_5.5 and SVL simulator fails.

Addi2020, 2021-07-23:
I am trying to connect apollo5.5 with SVL simulator but  I do not understand where I faild or miss something.

I use the following 2 files.
1. https://www.svlsimulator.com/docs/running-simulations/running-simulator/ 
2. https://www.svlsimulator.com/docs/system-under-test/apollo5-0-instructions/

add the following rules:
sudo ufw allow 8888
sudo ufw allow 9090
 
Steps:
./docker/scripts/dev_start.sh -l -t apollo5.5
./docker/scripts/dev_into.sh
./apollo.sh build_opt_gpu
./scripts/bridge.sh
./scripts/bootstrap_lgsvl.sh

EricBoiseLGSVL, 2021-07-23:
@Addi2020 We support 5.0 and 6.0 but I don't think we support 5.5 out of the box.  @lemketron ?

Addi2020, 2021-07-26:
@EricBoiseLGSVL This screenshot  shows the bridge connection between Apollo_5.5 dreamview and SVL simulator. 
bridge status= connected
[image]

but there is an issue
1. Routing is successful but the car do not move. It stop at initial position. Autopilot does not work.
2. the info shows an error connection refused. As shown in screen short.

[image]

tajinder96, 2021-07-26:
Hi, I also wanted to know about the connection to Apollo v5.5. Like you mentioned, I think v5.5 is currently not supported. Is it difficult to setup this connection?

lemketron, 2021-07-29:
Apollo 5.0 upstream does not work with SVL Simulator. 
The required changes are in the [lgsvl fork](https://github.com/lgsvl/apollo-5.0).

Apollo 5.5 upstream does not have the changes for SVL Simulator, and there is no lgsvl fork of Apollo 5.5.

Apollo 6.0 and master [upstream](https://github.com/ApolloAuto/apollo/) do include the changes to support SVL Simulator and we recommend you choose one of those versions.  Apollo 5.5 will not work and 5.0 is very old at this point.

closed 2021-07-30 by Addi2020