lgsvl/simulator #1617: How to fetch and Visualize camera images in autoware.auto RVIZ

tecena, 2021-07-23:
Hi,
I am trying to fetch and visualize main camera images in autoware.auto RVIZ from simulator.
The LGSVL-Bridge and simulator are running on two different machines.
I did choose Lexus2016RXHybrid vehicle ,it has autoware.auto sensor configuration but does not have a camera on to it, so added a camera by clicking on + icon configurator and added color camera sensor, and matched the values with
{
  "type": "Color Camera",
  "name": "Main Camera",
  "params": {
    "Width": 1920,
    "Height": 1080,
    "Frequency": 15,
    "JpegQuality": 75,
    "FieldOfView": 50,
    "MinDistance": 0.1,
    "MaxDistance": 1000,
    "Topic": "/simulator/camera_node/image/compressed",
    "Frame": "camera"
  },
  "transform": {
    "x": 0,
    "y": 1.7,
    "z": -0.2,
    "pitch": 0,
    "yaw": 0,
    "roll": 0
  }
},
Configuration.
Created a Random Traffic simulation with Autowrae.Auto(ApexAI) autopilot with IP and Port of the machine where LGSVL-Bridge is running.
When I run simulation, the LGSVL bridge connected to simulator with IP:Port . The simulation is starting , bridge is connected at  the specified IP, data is flowing from simulator to LGSVL_bridge. The Ros topics listed are 
[image]

I am able to see the topic /simulator/camera_node/image/compressed.

I am able to add this topic RVIZ, but no image is displayed.
[image]

Even when I run- ros2 topic echo /simulator/camera_node/image/compressed.
No data is displayed.

I did download the image transport package - sudo apt install ros-foxy-compressed-image-transport 
and also executed the image decompressor and publisher command - 
1.ros2 run image_transport republish compressed in/compressed:=/simulator/camera_node/image/compressed raw out:=/image_raw
2.ros2 run image_transport republish compressed in:=/simulator/camera_node/image raw out:=/new_topic_name

also referred the Autoware video lecture 11 for the same https://www.youtube.com/watch?v=OcB6FUbjZXo, its working correctly in video , but when I follow the same ,its not working out

Please help me to know, what has to be done to fetch and visualize main camera data into Autoware.Auto RVIZ.

tecena, 2021-07-28:
@cyberphysicalrobotics 
please help to fetch the camera topic images .

mguzzina, 2021-09-13:
Could you try running `ros2 run image_transport republish compressed /in/compressed:=/simulator/camera_node/image/compressed raw /out:=/image_raw` instead?