lgsvl/simulator #1618: Storing ROS TOPIC data even after the scenario execution is completed from python-api

AKSHATA-MK, 2021-07-23:
Hi,

I am running a scenario python-api script with Lexus2016RXHybrid vehicle with Autoware.Auto sensor configuration, connected  LGSVL_BRIDGE to simulator using IP address. I am getting ROS topics in LGSVL bridge, But this data is available only until the scenario python script is getting executed in the simulator, once script execution is completed, no ros topics/no data is available,
How do I get/store the ROS TOPIC data even after the scenario python script execution is completed.

Please help

Thanks

mslavescu, 2021-07-24:
@AKSHATA-MK you can record a ROS bag while the simulation is running, and play it back and view the info after the simulation ended.

For ROS check this page:
http://wiki.ros.org/ROS/Tutorials/Recording%20and%20playing%20back%20data

For ROS2 check this page:
https://roboticsbackend.com/ros2-bag-save-and-replay-topic-data/

AKSHATA-MK, 2021-07-26:
> @AKSHATA-MK you can record a ROS bag while the simulation is running, and play it back and view the info after the simulation ended.
>
> For ROS check this page: [...]

> @AKSHATA-MK you can record a ROS bag while the simulation is running, and play it back and view the info after the simulation ended.
>
> For ROS check this page: [...]

@mslavescu Thank you for the reply.
We have already  tried recording into a .db3 file using- ros2 bag record <topicname>
We are mainly focused on camera image data. But in the .db3 file generated, the camera image data is not in image format,  

so we would want to know is there any API available to fetch the camera data directly from .db3  file or any other way to fetch the camera image directly.

mslavescu, 2021-07-27:
@AKSHATA-MK I was able to record ROS2 bags with images from SVL camera sensors.

Can you share your SVL vehicle sensor config file? 
You may have some issues with topics.

AKSHATA-MK, 2021-07-28:
> @AKSHATA-MK I was able to record ROS2 bags with images from SVL camera sensors.
> 
> Can you share your SVL vehicle sensor config file?
> You may have some issues with topics.

Here is my sensor configuration file.

[
  {
    "params": {
      "Topic": "/lgsvl/vehicle_state_cmd"
    },
    "name": "Autoware Auto Vehicle State",
    "parent": null,
    "pluginId": "8aff00f2-a5e4-4bd3-a778-517f9307fa99",
    "plugin": {
      "assetGuid": null,
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "supportedSimulatorVersions": [
        "2021.1",
        "2021.1.1",
        "2021.2"
      ],
      "id": "8aff00f2-a5e4-4bd3-a778-517f9307fa99",
      "name": "Vehicle State Sensor",
      "type": "VehicleStateSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "Example of a sensor to get state of different parts of vehicle.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      },
      "shareRequests": []
    },
    "type": "VehicleStateSensor"
  },
  {
    "params": {
      "Frequency": 10,
      "Topic": "/lgsvl/state_report"
    },
    "transform": {
      "x": 0,
      "y": 0,
      "z": 0,
      "pitch": 0,
      "yaw": 0,
      "roll": 0
    },
    "name": "CAN Bus",
    "parent": null,
    "pluginId": "b0bdc474-ac09-4355-901d-d7012a8c57a8",
    "plugin": {
      "assetGuid": null,
[... 484 lines cut ...]
      "supportedSimulatorVersions": [
        "2021.1",
        "2021.1.1",
        "2021.2"
      ],
      "id": "483f7b90-2f76-42ee-82c1-22f02f25f924",
      "name": "GPS Odometry Sensor",
      "type": "GpsOdometrySensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This sensor outputs the GPS location of the vehicle in Longitude/Latitude and Northing/Easting coordintates and the vehicle velocity.\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#gps-odometry for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      },
      "shareRequests": []
    },
    "type": "GpsOdometrySensor"
  }
]