lgsvl/simulator #1620: When export HD Map (autoware vector map),some csv is empty

windwolf9, 2021-07-24:
Hello @EricBoiseLGSVL,
When i export by(Unity: Simulator->Export HD Map->(select: Autoware Vector Map)),and the signaldata.csv,vetor.csv,pole.csv is
empty,but other is not。 Can you tell me the reason ,thanks!
[image]

[image]

windwolf9, 2021-07-26:
I use :simulator-2020.06  unity version 2019.3.15f1

EricBoiseLGSVL, 2021-07-26:
please see #1533

windwolf9, 2021-07-27:
Thank you for your reply! it  means that the latest version can export data normally? But when i use simulator-2021.2，it also can not export signaldata.csv,vector.csv,pole.csv. It seems the function of  Export HD Map(Autoware Vector Map) has some problem ?

EricBoiseLGSVL, 2021-07-27:
hmm, it should but it could be bugged.  We'll look into it.

closed 2021-07-27 by EricBoiseLGSVL