lgsvl/simulator #1621: Highway101GLE does not show correcty in DreamView

GabrielZZZ, 2021-07-24:
After downloading the map from the [Store](https://wise.svlsimulator.com/maps/profile/431292c2-f6f6-4f5a-ae62-0964f6018d20) and using the map generation techniques from doc and here #1521 , and following the steps to restart the DreamView, the Highway101GLE map does not show correctly in the latest Apollo system.

Apollo Version: Master
System: Ubuntu 20.04

Screenshots:
- Files in the map folder
[image]

- DreamView:
[image]

lemketron, 2021-07-24:
What does the Module Controller tab look like (are Transform and Localization enabled)? 
What are the details for your simulation? Is it configured for Random Traffic? 
You're using Apollo master? 
With "modular testing" (ground truth sensors) or Apollo perception?

GabrielZZZ, 2021-07-25:
HI @lemketron .

Yes. I am using the Apollo master with modular testing. Here is the screenshot for the sensors in DreamView and simulation configuration:
[image]
[image]

> What are the details for your simulation? Is it configured for Random Traffic?
It is not configured for Random traffic. I am customizing the scenario to test a simple traffic jam situation:
[image]

I have used the same methods for the map Straight2LanseSame, and it works fine. So I wonder is there something wrong with the base map file of the Highway101GLE?

revati-naik, 2021-07-25:
@GabrielZZZ Could you please share which simulator version are you working on?

GabrielZZZ, 2021-07-26:
@revati-naik Yes. It is version 2021.2 (latest)

OYhhh2, 2021-08-13:
I have the same problem,I have tried lots of may , the Sanfranciso and Shalun also show incorrect in dreamview

revati-naik, 2021-08-18:
@GabrielZZZ I just verified that the map does not show up in Dreamview. I will take this issue with the team and get back to you. Thanks for bringing this up.