lgsvl/simulator #1623: Python API using wise.preview server

risk-v, 2021-07-26:
Hi everyone,

I was wondering if it's possible to run a simulation that uses the Python API and Autoware.AI v.1.14   on the development server (wise.preview.svlsimulator). The simulator (inside Unity) says API Ready but when I execute the python script nothing happens. Thanks in advance.

EricBoiseLGSVL, 2021-07-26:
What does your player.log say from Unity? What python code are you running?  Are you running APIOnly template?

lemketron, 2021-07-26:
What output do you get from running the Python script? Sounds like it might not be connecting, but the console output from Python along with the Simulator's player.log file should help figure out what's happening (or not happening).

risk-v, 2021-07-27:
@EricBoiseLGSVL , @lemketron Thanks very much for your suggestions. The Python API is working now (e.g. running example #04 in /quickstart). But the bridge is still not working. I realized that an Apollo compatible car is being downloaded (Lincoln2017) even though I specified Lexus2016RXHybrid in the Simulator set up window (I tried the Jaguar too, but, same result: keeps downloading and using the Lincoln car). Because of that Autoware.AI can't establish a proper communication. Am I missing something? Thanks in advance.

EricBoiseLGSVL, 2021-07-27:
- I suggest not using preview for development.  It is not stable.
- In the python script are you referencing the vehicle sensor config ID or using the default vehicle set in the python api?  If there is a bug in the python api template, I recommend just running the ApiOnly template instead.  There you just set the vehicle sensor config ID and it will load the correct one.

risk-v, 2021-07-28:
Hi @EricBoiseLGSVL,

Thanks for your answer. Great! Now it's working. I wasn't using the right ID. The only issue now is that some of the provided .py scripts that require NPC don't work because the assets seem to be missing even though I installed them in ./External/NPCs. I have attached a couple of screenshots:

This is the error I get when I run "10-npc-follow-the-lane.py". It complains about the NPC "Sedan":

[image]

This is how the Assets folder looks (Sedan installed):

[image]

I have to say that the preview mode has been very stable for me. I need that particular mode/flexibility because I am creating my own extremely custom sensors :) Thanks in advance.

EricBoiseLGSVL, 2021-07-28:
I understand preview is good when you need flexibility but just be aware the database can be wiped at anytime and what is pushed might have bugs that are currently being worked on.
Simulator no longer uses local prefabs for pedestrians and npcs.  They need to be built and placed in AssetBundles folder.
I see that you didn't clone the default npcs correctly. See docs [here](https://www.svlsimulator.com/docs/plugins/npc-plugins/).

[image]

[image]

[image]

risk-v, 2021-07-29:
Hi @EricBoiseLGSVL,

Thanks. I will bear that in mind when using the preview mode. Regarding the NPCs I followed the instructions from the link you provided but I am still getting the same error:

[image]

I tried many different things like building the Sedan NPC many times, refreshing, different folders structures, opening/ closing Unity again, compiling the source instead of using the Unity Editor, copy/pasting the AssetsBundles NPC folder from the "2021.2" release, but nothing worked. I realized that when I build a Sedan NPC, for instance, I end up with the "npc_Sedan" that is only 26 Kb instead of Mb. Do I have to do something else with the Python API? (e.g. Update a setup file...). Thanks in advance.

EricBoiseLGSVL, 2021-07-29:
Yes, if you have it at 26kb then something is not right.  If you download the latest release, these NPCs are already built.  Just add to the AssetBundles folder.  If you want to build locally, you need to be sure git lfs is installed and you have both linux and windows mono support.  Please look at you console log when building the bundle.  Nothing needs done on the python side, unless you haven't pulled the latest.  It pulls from name lookup in the prefab.
- install git lfs
- clone DefaultNPC to Assets/External/NPCs
- build bundle linux will be slightly larger than the size I posted
- if running from editor, bundle will be placed in the right folder to run automatically.
- if running in a binary, place the asset in AssetBundles/NPCs/DefaultNPC/

risk-v, 2021-07-30:
Hi @EricBoiseLGSVL,

First of all. Thanks for taking the time to reply to my previous comments and for the great help. I followed your previous steps. but I am still getting the same error I shared in the last post. I have "git lfs" up and running and I am cloning DefaultNPC from here -> https://github.com/lgsvl/DefaultNPC

When I build the NPCs assets, all the NPCs (e.g. Sedan) are around 26kb. I thought that was the problem, so I tried pedestrians instead of NPCs. After downloading "Walkers" from https://github.com/lgsvl/Walkers and compiling the character "Bob" is around 4 Mb which is great news, but then when I run the script "15-pedestrian-walk-randomly.py" I get the same "reference not found" error but this time pointing "Bob" as the problem. When I run that Python script the Unity console says there is a WebSocket issue:

[image]

Just a reminder, scripts that don't use NPCs or pedestrians work perfectly. I tried to download the 2021.2 source again and reinstalling the Python API, but got the same errors. I created different clusters/simulations on the preview server but still got the same error. I also compiled the source and used the non-preview website but same errors. The pre-built version works perfectly so obviously there is something wrong while I am compiling from source. The problem with the pre-built version is that it can't use some of my custom sensors. Looks like a reference issue when recompiling and Python is not accessing the right .json file or something. Any ideas? Thanks in advance.

EricBoiseLGSVL, 2021-07-30:
No problem, glad to help.

This header issue is fixed in the next release but it usually is only when the api stops.
When you are in editor and first open it are you seeing this log?
[image]

This tells you that the files are loaded successfully.  I'm not seeing this in your screen shots.
Yes, a 26kb NPC is a problem.  That means the textures and other data is not there.  Please attach the npc you are building.

Also screen shot the folder you are placing the npc and the folder you are placing the npcs in the editor. Also, screen shot the texture in the sedan folder.

Have you tried to use the npcs and pedestrians that are included in the binary release?  Why are you building them again?

risk-v, 2021-07-31:
Hi @EricBoiseLGSVL,

Thanks again. My log seems correct. The compilation assets size is fixed now (it was a Git Lfs issue). Here you have a screenshot:

[image]

I can confirm now that my assets are being compiled and recognized properly by Unity. But they are still not being recognized by the Python API.  Still getting that same error when I try to run the code "10-npc-follow-the-lane.py" or any other provided example scripts that use NPCs or pedestrians:

[image]

Regarding your last question, the reason why I am trying to build my own NPC assets it's because I was getting that same error when using the default (pre-built) ones and I thought that recompiling them will fix the issue. I thought it was something related to using different versions of Unity, ids not matching with my build or something like that, but obviously, I was wrong :)

Here you have the screenshots you were asking for:

1) This is the NPC Sedan folder inside the Editor:

[image]

2) And this the folder where the built assets are placed:

[image]

Once I have the assets build, I would normally start the simulation in the browser (in my case I am using the wise.preview), wait for "API Ready!",  then I open another terminal go to ../quickstart/ and run my scripts with the following command "python3 10-npc-follow-the-lane.py" and that's where I get the shown error. If the script doesn't contain NPc or pedestrians it works perfectly.

I was looking at the Python API source but I am not sure where it goes to find a reference to the Sedan asset. I am starting to believe it might be an internal problem or something really obvious that I missed during the installation :) But I think I followed all the steps carefully. Any ideas? Thanks in advance.

EricBoiseLGSVL, 2021-08-02:
Thanks for the images and explanation, this helps.  I see why you are building locally and it's great that you resolved the build issue.
Now for the api issue.
- This all looks correct. Unity is logging that the npcs have been loaded. The api looks for these in memory when running a api script and should work fine.  The only thing I can think of is that you have an older version of Python API.  What tag is your python repo on?

EricBoiseLGSVL, 2021-08-03:
@revati-naik please file a bug with python api template when you are able.
@risk-v glad you got it working, sorry about the issues with python api template

EricBoiseLGSVL, 2021-08-05:
@risk-v we have found the bug with NPC names as well as pedestrians.  Here is the patch 
[APINPCPedpatch.zip](https://github.com/lgsvl/simulator/files/6941001/APINPCPedpatch.zip)
and we will be pushing the fix next release.  Thanks for the issue.

closed 2021-08-05 by EricBoiseLGSVL