lgsvl/simulator #1625: How to create a camera sensor with 120 degree of Horizontal FOV

joeking11829, 2021-07-27:
Hi LGSVL Teams,

the configuration of camera sensor:
width: 1280
height: 960
vfov(vertical fov): 90.0
hfov(hotizontal fov): 106.26020470831195

Because the maximum fov(vertical fov) of camera sensor is 90 degree.
How could I create a camera sensor with 120 degree of HFOV.

Any ideas ?
Thanks.

EricBoiseLGSVL, 2021-07-27:
Others have done this with cubemap camera options. @kamkolak-lge ?

kamkolak-lge, 2021-07-29:
It should be possible to achieve higher FoV through fisheye camera, but that would also introduce some fisheye-specific distortions.

The 90 degree maximum is an arbitrary value, so it should be possible to simply increase maximum value in the sensor code. You could try modifying maximum `FieldOfView` value in [`CameraSensorBase`](https://github.com/lgsvl/simulator/blob/master/Assets/Scripts/Sensors/CameraSensorBase.cs) and rebuilding [`ColorCameraSensor`](https://github.com/lgsvl/ColorCameraSensor). WISE should then reflect the change in parameter range and allow you to set it.

joeking11829, 2021-08-03:
> The 90 degree maximum is an arbitrary value, so it should be possible to simply increase maximum value in the sensor code. You could try modifying maximum FieldOfView value in CameraSensorBase and [...]

Thanks.
I will try it.

BTW, how to calculate the FOV of fisheye camera?

kamkolak-lge, 2021-08-13:
> BTW, how to calculate the FOV of fisheye camera?

Given fisheye camera with field of view `f`, center of projection offset `Xi`, and all plumb bob distortion coefficients all set to 0, effective field of view `f'` after the distortion can be calculated with this formula:
```
f' = f + 2 * asin(Xi * sin(0.5 * f ))
```

joeking11829, 2021-09-09:
> Given fisheye camera with field of view `f`, center of projection offset `Xi`, and all plumb bob distortion coefficients all set to 0, effective field of view `f'` after the distortion can be [...]

Got it.
Thank you.

closed 2021-09-09 by joeking11829

peter0617ku, 2022-03-30:
@kamkolak-lge 

Hello, sorry to disturb you!
I want to use the FOV of the fisheye to calculate the focal length of the fisheye, but I have some difficulties.
May I know the origin of this formula?

Thank you for taking the time out of your busy schedule to help!