lgsvl/simulator #1630: Color camera SensorDistributionType set to Mainonly, it fails to start the simulation.

deepalcoholic, 2021-07-27:
when the Color camera SensorDistributionType set to Mainonly, 
it fails to start the simulation.
Would anyone can help?

EricBoiseLGSVL, 2021-07-27:
- What version of simulator are you on?
- What OS are you on?
- What errors are you getting from player.log?

deepalcoholic, 2021-07-28:
> * What version of simulator are you on?
> * What OS are you on?
> * What errors are you getting from player.log?

@EricBoiseLGSVL 
version:2021.21.
OS: Ubuntu18.04
log: 
[Player.log](https://github.com/lgsvl/simulator/files/6892108/Player.log)

deepalcoholic, 2021-07-28:
> * What version of simulator are you on?
> * What OS are you on?
> * What errors are you getting from player.log?

@EricBoiseLGSVL  
version: 2021.2.1
OS: Ubuntu18.04

closed 2021-07-28 by deepalcoholic

EricBoiseLGSVL, 2021-07-28:
Did you find a solution to the issue?

Just1045523143, 2022-07-23:
@EricBoiseLGSVL Do you have any ideas? i also meet this question. ubuntu20.04, lgsvl 2021.3 and 2021.2.1 both try.