lgsvl/simulator #1631: How to make autoware.auto control the ego-vehicle

tecena, 2021-07-27:
Hi,
We have setup autoware.auto to run along with SVL simulator. We are able to fetch ros2 topics data from simulator onto lgsvl-bridge. Here the ego-vehicle is controlled by the PythonApi script.

But how can we make the autoware.auto to control the ego vehicle instead of python script. Like if we give start and end points in python script, the vehicle should be driven by autoware.auto along the direction.

we did follow a documentation to achieve it https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html.
I am able to launch the simulator with **/opt/lgsvl/simulator** command. but the simulator is stuck at
[image]

Even though I can open the wise webapp in browser with local host http://127.0.0.1:8080/ how can I create a simulation without a cluster.

How can we control the  ego-vehicle  by autoware.auto by specifying start and end points in the python script.

Please help

EricBoiseLGSVL, 2021-07-27:
We do not support creating a simulation without creating a cluster.  Is that what you are trying to do? This is also a very old version of the simulator, I recommend updating to the latest version. @hadiTab ?

tecena, 2021-07-28:
> We do not support creating a simulation without creating a cluster. Is that what you are trying to do? This is also a very old version of the simulator, I recommend updating to the latest version. @hadiTab ?

@EricBoiseLGSVL Thank you for the reply
We will consider upgrading the simulator.

No I am  not trying to create a simulation without a cluster, we are using the api-only mode of simulation, so we want to make the ego vehicle to be controlled by autoware.auto  instead of python script.Like if we Just mention starting and end point in the script the vehicle should be driven by autoware.auto through out the direction.

To achieve this we had followed the documentation
https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html
In this step, once I launch the simulator inside ade, it is stuck at the screen I shared above, open in browser is not clickable,so I can not create a cluster for the simulation

How do we achieve ego-vehicle controlled by autoware.auto
Please help

EricBoiseLGSVL, 2021-07-28:
Ok I see, the error is the webui won't open.  Did you install npm correctly? NodeJS? Again, this is a very old version of simulator and we really don't support it.  I recommend updating to release-2021.2.1.

You need to post the player.log for simulator to see what the errors are.

Autoware foundation docs are very outdated.  Have you looked at the latest from us yet? [Here](https://www.svlsimulator.com/docs/system-under-test/autoware-auto-instructions/).

tecena, 2021-07-30:
> Ok I see, the error is the webui won't open. Did you install npm correctly? NodeJS? Again, this is a very old version of simulator and we really don't support it. I recommend updating to [...]
@EricBoiseLGSVL actually  the simulator that we have downloaded in adehome folder is latest version is 2021.1. I have downloaded a xip file and extracted it. When I launch this simulator by double clicking the latest simulator window opens and all buttons are functional, able to open wise in browser can create simulations and run them.
We have not downloaded any older versions of simulator.

But when I try to run launch the simulation inside ade with **/opt/lgsvl/simulator** command this older version of simulator is launched, I am not sure how this version lgsvl image is created inside ade inspite of the fact we have not installed any older version of simulator.

I tried to still update (delete and download)the newer version 2021.2.1 of simulator inside adehome folder, but still this is not getting updated inside ade it still launches the older version,

How do I update the simulator inside ade.

Please let me know if I am doing anything wrong

EricBoiseLGSVL, 2021-07-30:
Sorry I am not as familiar with ade. Any idea @lemketron or @hadiTab ?

mslavescu, 2021-08-01:
@tecena keep 2020.06 on that line, SVL 2021 versions are not supported yet in Autoware.Auto.
Run ADE same as for LGSVL 2020.06 and then run SVL 2021.2 (latest is 2021.2.1) outside of ADE docker.

If you run SVL on a different machine than where the ADE/Autoware.Auto is running you can use the IP of the ADE machine in the ROS2 bridge connection in SVL simulation Autopilot page.

To use Autoware.Auto to drive the car you'll need the lgsvl_bridge running in ADE docker, check this comment and issue for more details:

https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/977#note_611627609