lgsvl/simulator #1632: SVL fork Apollo 5.0: not able to run bridge

tajinder96, 2021-07-27:
Hi, 

I am trying to work with the Apollo 5.0 fork of Lgsvl, mainly to use the camera-based object detection in Apollo. When I run bridge.sh after building Apollo, I get the following error "cyber_bridge: no such file or directory". Do you know what I can do to solve this? 

I also have another question. Will having a different version of Apollo (say Apollo 6.0) also built on the same computer cause conflicts between the two versions? Seems like this is the case for native Apollo versions [,](https://github.com/ApolloAuto/apollo/issues/11272) but not sure if this is true for the Lgsvl fork as well. 

Thanks a lot!

EricBoiseLGSVL, 2021-07-27:
- The error is stating it.  The bridge.sh is expecting the cyber_bridge in a certain file or directory. Can you give more details to what you are trying to do or what you have tried so far? What version of simulator are you on?

- I am guessing that if this is an issue on native then yes it will cause an issue with the fork. @lemketron ?

tajinder96, 2021-07-28:
Hi, I am using v2021.2 of SVL. I simply followed instructions on the installation as here [apollo5_instructions](https://www.svlsimulator.com/docs/archive/2020.06/apollo5-0-instructions/), then ran the bootstrap script, and then tried to run bridge.sh, but encountered this error. (seems like the document is an archived one, but i compared it to the new documentation - don't think anything changed)

EricBoiseLGSVL, 2021-07-28:
Good so you are on 2021.2.  Now what commands are you using exactly?  Do not use the archived steps if new docs have been posted.  Have you tried rebuilding Apollo?  Sometimes this is the best way to fix issues with it.  More information is needed to help you setup Apollo. @lemketron ?

tajinder96, 2021-07-29:
Thanks Eric. Actually, I am trying something else right now with Apollo-master, and I don't want to try to build two different versions of Apollo on the same computer at this moment (as I mentioned, there may be conflicts). I will follow up with you on this a bit later on. 

Just wanted to check on a related matter. Do you by any chance know if recording simulator data using the bridge in Apollo v6.0 as CyberRT messages and then replaying the data to run open-loop simulation with Apollo 5.5 / Apollo 5.0 will work well? Thanks!

EricBoiseLGSVL, 2021-07-29:
@tajinder96 I would ping someone on Apollo issues since we haven't tried that scenario yet.  Please let us know what you find though, sounds interesting.

Addi2020, 2021-07-30:
@EricBoiseLGSVL I am also working on the same issue. I also use Apollo-5.0 fork for creating bridge connection between apollo and Lgsvl, but gives an error. 
https://github.com/ApolloAuto/apollo/issues/13995

Is it due to conflict between apollo_5.5 and apollo_5.0 fork? as mentioned by @tajinder96 
How to resolve this issue?

EricBoiseLGSVL, 2021-07-30:
1) more than likely.  It seems multiple versions of apollo on one machine cause issues
2) If you don't require perception, use 6.  We have modular support to work around a bug in Apollo.  If not, use 5 and use a clean build of apollo

tajinder96, 2021-08-01:
I'll close the issue for now, and re-open when needed. Thanks everyone!

closed 2021-08-01 by tajinder96