lgsvl/simulator #1633: Interactive information between simulator and cloud

DZiYou, 2021-07-28:
I would like to ask what information is exchanged between the local simulator and the cloud, and whether it is possible to separate from the cloud for stand-alone debugging. Because I don't have an external network here. What source code do I need to modify if I don't want to use the Internet? 
Or is there an interface configuration for stand-alone debugging?
thank you

EricBoiseLGSVL, 2021-07-29:
We have offline simulations but it requires a network connection to run once first.  @zelenkovsky

DZiYou, 2021-07-29:
I noticed the offline simulation mode. However, only the simulation edited in online mode can be run in offline mode. That is it means I can't edit the simulation in offline mode? Select a other map, etc. 
These can only be in online mode, right?

EricBoiseLGSVL, 2021-07-29:
Yes that is correct.  We are working on development settings to run from a scene but that still requires an internet connection to pull sensor configurations.