lgsvl/simulator #1634: Asset Bundle uploads failed

kruztw, 2021-07-28:
I know this issue is duplicated, but many of them are closed without solutions.

Error:
[image]

Asset Bundle is downloaded from:
https://content.lgsvlsimulator.com/maps/cubetown/

Any advice is appreciated.

EricBoiseLGSVL, 2021-07-29:
What cloud url are you using for these 06 bundles?

kruztw, 2021-07-29:
@EricBoiseLGSVL 
https://wise.svlsimulator.com/maps/profile/add

kruztw, 2021-07-29:
If I could, I would like to run on localhost, like [this video](https://www.svlsimulator.com/docs/simulation-content/add-new-map/) at 6:18.
But it seems link the recent versions of SVL simulator can only link to the cloud.

EricBoiseLGSVL, 2021-07-29:
- The 2020.06 maps are not compatible with wise.svlsimulator.com. This is a very old version of simulator.  If you want to use these assets you need to use the 06 release binary or the release tag for editor.
- This is the same step in the latest as in the video.  You need to open the LoaderScene in Unity and hit play.  It will behave just like the video and run like a binary in the Editor.
- I recommend you move to the latest release 2021.2.1 and unity 2020.3.3f1 as per our docs [here](https://www.svlsimulator.com/docs/getting-started/introduction/).

kruztw, 2021-07-29:
@EricBoiseLGSVL 

Let me clear something.
wise.svlsimulator.com is for 2021.x (included 2021.1 and 2021.2)
The [assetBundle link](https://content.lgsvlsimulator.com/maps/cubetown/) is for 2020.06 which is not compatiable with wise.svlsimulator.com.
Is that right ?

However, I use latest-2021.1 simulator and unity 2020.3.3f1.
[image]

And build the assetBundle, then upload to the cloud.
It still complains the same error.
[image]

I don't know why, maybe l should try latest-2021.2

EricBoiseLGSVL, 2021-07-29:
- Yes, 06 is not compatible.
- You have two annotation holders in the scene.  Please delete one that is not needed and build again.

kruztw, 2021-07-29:
@EricBoiseLGSVL 

I delete the MapShalun (1) and rebuild, but it still pops the same error when upload to cloud.
So I try to use 2021.2.1
[image]

And add it in Unity hub with unity 2020.3.3f1
[image]

And the error occur
[image]

I click ignore
[image]

and there is no simulator in the bar.

EricBoiseLGSVL, 2021-07-29:
this means you have errors in the scripts and Unity can't compile. Be sure all your repos that you have cloned are at the same release tag.
Also, since you have been switching unity versions, you may have corrupted Unity's Library files. (cached shaders).  If all else fails:
1) Close unity
2) Delete the Library folder in the unity root folder of the project
3) Open Unity, let it recompile the shaders.
4) Close Unity
5) Open Unity again.  You may see cyan shaders on meshes when you open scenes/prefabs. this is fine, let Unity recompile the shaders before building again.

kruztw, 2021-07-30:
@EricBoiseLGSVL 
Thanks for your help, the unity works great !

For those who met the same problem :

the following commands are not recommend
**git clone https://github.com/lgsvl/simulator  
git checkout 2021.2.1**

I removed the older simulator, and 
**git clone https://github.com/lgsvl/simulator --branch release-2021.2 --single-branch**

the errors were all disappeared.
( I cloned cube Town in the Assets/External/Environments )
[image]

However after building the Asset, and upload it to the cloud, the error pops up.

Firefox:
[image]

Chrome:
[image]

kruztw, 2021-07-30:
more detail:
[image]

EricBoiseLGSVL, 2021-07-30:
Hmm, @thisisdavidfrank what could cause the Gateway timeout?  @heeen was this fixed with the websocket sharp update?

kruztw, 2021-08-02:
@EricBoiseLGSVL
Is therer any progress ?

heeen, 2021-08-02:
@EricBoiseLGSVL websocketsharp is only used by API mode and ROS bridge

heeen, 2021-08-02:
It appears your asset bundle was not built correctly, as it should unzip without errors. I assume that the wise site also has issues unpacking it. can you try to rebuild it and look at the console window after the build finishes to check for errors from the build process?

heeen, 2021-08-02:
After a successful build a popup like this should appear:
[image]

there should not be errors from unzipping
```
$ mkdir tmp
$ cd tmp/
$ unzip ../environment_CubeTown 
Archive:  ../environment_CubeTown
 extracting: hdmaps/lanelet2/CubeTown.osm  
 extracting: hdmaps/opendrive/CubeTown.xodr  
 extracting: hdmaps/apollo50/base_map.bin  
 extracting: hdmaps/apollo30/base_map.bin  
 extracting: hdmaps/autoware/AutowareVectorMap.zip  
 extracting: images/preview-1.png    
 extracting: 5e793b6a-4f4d-4fd3-ad83-b4f89b5f614f_environment_main_linux  
 extracting: 5e793b6a-4f4d-4fd3-ad83-b4f89b5f614f_environment_textures  
 extracting: 5e793b6a-4f4d-4fd3-ad83-b4f89b5f614f_environment_main_windows  
 extracting: manifest.json           
```

@eagafonovlge can you check  if broken zip files could cause above error messages?

kruztw, 2021-08-02:
@heeen 
Looks like it lacks of Linux Module.
[image]

heeen, 2021-08-02:
Install it from unity hub
[image]
[image]

kruztw, 2021-08-02:
It works ! Thanks for all your helps.

closed 2021-08-02 by kruztw