lgsvl/simulator #1636: Exception: Simulator is not linked. Enter play mode to re-link.

kruztw, 2021-07-28:
I followed this guide, and used branch 2021.1
https://www.svlsimulator.com/docs/running-simulations/developer-mode/

But when I clicked play button, the error popped up.
[image]

Please help. Thanks!

kruztw, 2021-07-28:
This is what I expected
[image]

dogukannn, 2021-07-28:
https://github.com/lgsvl/simulator/issues/1446#issuecomment-843656375

kruztw, 2021-07-28:
Thanks for your help, the problem is solved.

For those who met the same issue, click File -> Open Scene -> select simulator\Assets\Scenes\LoaderScene.unity -> Open, then it is good to go.

closed 2021-07-28 by kruztw