lgsvl/simulator #1637: The video scene is not precisely synchronized within the Main and the slave client on one Cluster running

deepalcoholic, 2021-07-28:
Version: 2021.2.1
OS: Ubuntu 18.04
The video is not precisely synchronized within the Main and the slave clients on one Cluster running.
Is this a bug?

EricBoiseLGSVL, 2021-07-29:
Which video? From the video recording sensor or from the main simulator camera? If the simulator camera then yes, there is latency in the simulator camera.  This is not sensor data (which is synchronized with a clock sensor) and is expected.  @KarByc

KarByc, 2021-07-29:
Thanks for using the SVL Simulator with distributed simulation.
Please attach the video, we will review synchronization in your case.
Distributed simulation expects minimum 2 frames delay (~0,03s) in the local wired network with machines running simulations in 60fps.
Simulations' cameras and not synchronized between the cluster machines.