lgsvl/simulator #1640: ArgumentNullException: Value cannot be null.

kruztw, 2021-07-28:
I want to create a new map, so I follow this video.
https://www.svlsimulator.com/docs/simulation-content/add-new-map/

However, modern city doesn't exist, so I cloned Shalun and Jaguar2015XE to Environments and Vehicles folders respectively.
And I clicked play button, but some errors popped up.

[image]
[image]

Any advice is appreciated.

EricBoiseLGSVL, 2021-07-29:
You cannot run the simulator from the scene.  You need to upload the bundle to the webui to run a simulation.  You need to start the sim in editor from the LoaderScene.  We are working on a way to run from scenes in Developer Settings but it is still a wip and has issues.  We hope to release a fix soon.

kruztw, 2021-07-29:
Thanks for your reply.
But I still confuse how the video works at 3:09, it just hit play button without using LoaderScene.

kruztw, 2021-07-29:
@EricBoiseLGSVL

kruztw, 2021-07-29:
Oh, I think I got something.
The video omits the upload part.
It's not kind for newbie ...

closed 2021-07-29 by kruztw

EricBoiseLGSVL, 2021-07-29:
- Again, this is developer settings in the UI menu, hitting play is not omitting an upload but runs from the scene.  Now that the simulator requires cloud connection, the developer settings needs a refactor to work with the new backend. Please see my response in the other issue you posted.
- Sorry about the confusion, remaking videos is very time consuming and we try to update once changes are critical.
If you are just starting, I recommend you use the latest release before jumping into development mode. You should familiarize yourself with the code base and how to work with Unity.  Still, if you are already running sims in editor you are doing great so far.  Soon you will be an expert.

kruztw, 2021-07-30:
Thank you for your warm-heart ^^