lgsvl/simulator #1641: Importing real world map to the simulator

AmirTavakkoli, 2021-07-29:
Hi,

I am trying to import a real world map into the simulator. I am not sure if LGSVL supports the OSM. If it does, do I need to import the map to unity and then export it as Apollo 5.0 HD Map, Autoware Vector Map, Lanelet2,or OpenDrive 1.4? I appreciate it if I could have your help

EricBoiseLGSVL, 2021-07-29:
Sorry, we don't support OSM at the moment.  We are working on this though.
I would convert the OSM map into a format we support and then import and use our map mesh creation tool to create the roads.
Then, you can export from unity and do any cleanup, polish in a modeling tool.
We also support importing point clouds.

lemketron, 2021-07-29:
The approach would be to convert OSM to OpenDrive (although there's no simple turnkey way to do that just yet) and then import the OpenDrive HD map into SVL Simulator. As @EricBoiseLGSVL mentioned we are working on this since it would be a great way to add new maps into SVL.