lgsvl/simulator #1642: Error about build a new map

OYhhh2, 2021-07-29:
Hello, I got an opendrive map file and try to  add it to the simulator,so i follow this video
[image]
After i click the build button,it has some error,  However i don't understand what it means and how to fix it ,can you give some advice? (my english is not very good)
Unity edition:2020.3.3f1
[image]
[image]
[image]

EricBoiseLGSVL, 2021-07-29:
No worries and welcome to svl simulator.  This error is pointing to problems parsing the hd map. It points to the annotation object and let's you know as best it can what is the issue.
We use annotation for NPC navigation and for export to ADstack.  Sometimes the annotation is missing key data and we require users to alter the imported map to successfully run npcs and export.
The error I can see says that a stop line needs to have a public field in the component associated with a sign annotation.  You need to select that object and drag the closest or create a stop sign annotation reference.
The value cannot be null is usually an lane with no waypoints. (usually and error in the hdmap itself). We have checks for this, is this the latest simulator version? Are you on release 2021.2.1?
Hope this helps.  HDMaps and annotation is one of the most complex features of the simulator.

OYhhh2, 2021-07-30:
thank you very much ! Yes, I‘m using the latest simulator(2021.2.1),I have just uesd the simulator for three weeks,still a lot of work to learn

closed 2021-08-04 by OYhhh2