lgsvl/simulator #1643: Error about Start the bridge

enginlien, 2021-07-29:
Hello
When i was about to start the bridge,
this error message appeared.
[image]

shr-project, 2021-07-29:
Why are you running "dev_into.sh" when you're already inside the docker according to "danny@in_5_0_dev_docker" in PS1 ?

enginlien, 2021-07-29:
> Hello
> When i was about to start the bridge,
> this error message appeared.
> [image]

closed 2021-07-29 by enginlien

enginlien, 2021-07-29:
> Why are you running "dev_into.sh" when you're already inside the docker according to "danny@in_5_0_dev_docker" in PS1 ?

Thanks, I have revised the screenshot. 
[image]

lemketron, 2021-07-29:
It's not clear what "empty" channel bridge.sh is trying to create but you might see it if you make the window bigger.

Apollo 5.0 is getting pretty old at this point so unless you have a reason to be running something so old you should consider updating to Apollo 6.0 (or master) from [Apollo upstream on github](https://github.com/ApolloAuto/apollo/).

lemketron, 2021-07-29:
That said, it appears that this may be the same issue as #1359 where you are not specifying the `Topic` name in the sensor configuration: https://github.com/lgsvl/simulator/issues/1359#issuecomment-889381618

enginlien, 2021-07-30:
> That said, it appears that this may be the same issue as #1359 where you are not specifying the `Topic` name in the sensor configuration: [#1359 (comment)](https://github.com/lgsvl/simulator/issues/1359#issuecomment-889381618)

Hi Lemketron

Thanks for your reply.

I try to update to Apollo 6.0,
But the dream view still shows Please send car initial position and map data,
Could you please tell me why my dream always has this problem?

[image]