lgsvl/simulator #1645: Recorded images in the .db3 file is not getting displayed

zyan1234, 2021-07-29:
Hi ,

We are using  Lexus2016RXHybrid vehicle with autoware.auto sensor configuration, And added a camera sensor to capture camera data.
We are recording the ros topics using **ros2 record /simulator/camera_node/image/compressed** on the lgsvl-bridge. The .db3 file is getting generated. But when we try to convert the BLOB images data into images using the python script ,a number of images files are getting generated, but images are not displayed. We tried with multiple camera configurations, But still same issues persists.

This is our sensor configuration 

[
  {
    "params": {
      "Topic": "/lgsvl/clock"
    },
    "name": "Simulation Clock",
    "parent": null,
    "pluginId": "968f386f-dc0b-485d-ba33-5bb71bff93ef",
    "plugin": {
      "assetGuid": null,
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
          "lastName": "Content"
        }
      },
      "supportedSimulatorVersions": [
        "2021.1",
        "2021.1.1",
        "2021.2"
      ],
      "id": "968f386f-dc0b-485d-ba33-5bb71bff93ef",
      "name": "Clock Sensor",
      "type": "ClockSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "This sensor outputs simulated time to ROS as rosgraph_msgs/Clock message, or to CyberRT as clock message. The only parameter to use is topic/channel name.\nSee https://www.svlsimulator.com/docs/simulation-content/sensors-list/#clock for more details.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      },
      "shareRequests": []
    },
    "type": "ClockSensor"
  },
  {
    "params": {
      "Topic": "DepthCameraSensor",
      "Frame": "DepthCameraSensor"
    },
    "name": "Depth Camera Sensor",
    "parent": null,
    "pluginId": "290af702-b42a-4dce-800e-0999a07d49d2",
    "plugin": {
      "assetGuid": null,
      "isFavored": true,
      "isShared": false,
      "isOwned": false,
      "accessInfo": {
        "userAccessType": "favored",
        "owner": {
          "id": "0d888b00-fa53-47c1-882a-b68391268a11",
          "firstName": "SVL",
[... 531 lines cut ...]
        "2021.1.1",
        "2021.2"
      ],
      "id": "a5696615-d3e2-45e9-afa3-9b5ae7359e02",
      "name": "LGSVL Control Sensor",
      "type": "LGSVLControlSensor",
      "category": "sensor",
      "ownerId": "0d888b00-fa53-47c1-882a-b68391268a11",
      "accessType": "public",
      "description": "Example template of a sensor for vehicle Control information.",
      "copyright": "LG Electronics Inc.",
      "licenseName": "LG Content",
      "status": "active",
      "owner": {
        "id": "0d888b00-fa53-47c1-882a-b68391268a11",
        "firstName": "SVL",
        "lastName": "Content"
      },
      "shareRequests": []
    },
    "type": "LGSVLControlSensor"
  }
]
 
Please help

EricBoiseLGSVL, 2021-07-29:
This sound like an issue with your conversion code.  Can you use any other tool to look at the data that is saved?
The sensor config looks fine. Can you post images of the sensor running in simulator and a few images that are saved to compare?

zyan1234, 2021-07-30:
> This sound like an issue with your conversion code. Can you use any other tool to look at the data that is saved?
> The sensor config looks fine. Can you post images of the sensor running in simulator and a few images that are saved to compare?

@EricBoiseLGSVL 
I am sharing both images.One which is captured while simulation is running, The other is the image which is got from the .db3 file after conversion using our python script.
[image]

[image]

EricBoiseLGSVL, 2021-07-30:
Are you getting a db3 meta file?  I am seeing issues with extraction online from a lack of this file.  Other than that I don't see what the issue is.  It seems that the db3 file is messed up or the conversion code is not handling an edge case. is there any other tool you can debug with?