lgsvl/simulator #1646: SVL new Release fails to start simulation

marcusvinicius178, 2021-07-29:
Hi @EricBoiseLGSVL I am facing this error, after install the new SVL release: 2021.2.1 or also with the previous one 2021.1.1:

**lgsvl-simulator-scenarios-runner-simulator-build__2021.2.1.tar: no such file or directory**

I have already unlinked, cleaned the cache, created a new simulation from scratch as well as the clusters...

Could you please help me to figure out how to fix this issue?
Thanks.

EricBoiseLGSVL, 2021-07-29:
@revati-naik any idea what is happening? If there a step missing?

marcusvinicius178, 2021-07-29:
Thanks for the attention @EricBoiseLGSVL @revati-naik I have updated the question above...I also removed the vehicles and added again. But I am afraid that be a python issue...after I update...because any simulator version is working. I don't know how to recover the functionality

EricBoiseLGSVL, 2021-07-29:
have you updated python?

EricBoiseLGSVL, 2021-07-29:
Also adding @eagafonovlge

marcusvinicius178, 2021-07-29:
Yes I did. I have already purged python and restarted the installation, also from ubuntu-dekstop (I have purged and almost lost everything on my system hehehe).
But  I jsut checked the Python 3.8 is already the standard version on ubuntu 20.04. In other words this was not the issue, because it was working last week, the 2021.1 version. Then I checked now the docker images. And I got this:

[image]

I installed Apollo 6 today with the LGSVL tutorial: https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/

git clone https://github.com/ApolloAuto/apollo

I  am not sure why it dowloaded the docker image for the 2021.1 runner-scenarios

Shouldn'it take the newer releases?

The issue is that any release is working...

marcusvinicius178, 2021-07-29:
Hey @EricBoiseLGSVL an update....I tried again the new svl releases: 2021.2.1 and 2021.1.1 
and again the same issue of :

**lgsvl-simulator-scenarios-runner-simulator-build__2021.2.1.tar: no such file or directory**

However now the old simulator release started to work again...

I believe is something related to the docker image....would you know the how to dowload the docker images from the new releases...I mean could you share the cmd?

docker pull ....svl-2021.2.2 ???? 

Something like that could fix this issue I believe, and the scenarios could be built...I hope..

EricBoiseLGSVL, 2021-07-29:
We will look into this, thanks for posting @marcusvinicius178

lemketron, 2021-07-29:
> I installed Apollo 6 today with the LGSVL tutorial: https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/
> 
> git clone https://github.com/ApolloAuto/apollo
> 
> I am not sure why it dowloaded the docker image for the 2021.1 runner-scenarios

I believe you're seeing leftover docker images cached on your machine from having previously used the scenario runner in SVL Simulator. This is not related to apollo and should not affect it.

It sounds like you are changing simulator and/or apollo versions around, and it's not entirely clear which kind of simulation you are trying to run.  Which runtime template are you trying to use? Are you using `Python API` (which uses a built-in python scripts) or `API-only` mode? If you are trying to run a local python script with the simulator, then you'll need to use `API-only` mode in your simulation. This cannot be changed once you have created a simulation but you should be able to make a new `API-only` simulation. At that point the errors related to the runner container should go away.

marcusvinicius178, 2021-07-29:
Hi @lemketron  I am using the Python API . It runs with the old release, but not with the newer...
I runned with the standard code provided there, as well as mine, but the simulator does not start the simulation.

[image]

I was imagining that maybe some port or something code must be changed in Python Api code spot, however the error is the same: lgsvl-simulator-scenarios-runner-simulator-build__2021.2.1.tar: no such file or directory

Maybe I must restart the computer

> I believe you're seeing leftover docker images cached on your machine from having previously used the scenario runner in SVL Simulator. This is not related to apollo and should not affect it.

**Here is my doubt, should I remove this docker image, or clean the cache should fix?** Do you know the directory in which this cache is stored? If I clean it maybe solve the issue...I would like to try the new release, it seems LG provides new scenarios, right? This in fact is another doubt..**.I see a lot of nice new scenarios in SVL maps, but how to add them to Apollo or Autoware folders? Are these new maps' folders located in some repo?**
> It sounds like you are changing simulator and/or apollo versions around, and it's not entirely clear which kind of simulation you are trying to run. Which runtime template are you trying to use? Are [...]

marcusvinicius178, 2021-07-29:
In addition now that I tried the old release, it is not working anymore:

[image]

Last Time I needed to purge Python 3.8 totally from my computer to make it work....but I almost lost everything and needed to reinstall Ubuntu-Desktop. But the SVL worked afterwards.

I wouldn't like to redo this uninstallation...Is there other way to work around this issue?

EricBoiseLGSVL, 2021-07-29:
It seems you don't have access to the asset.  Is this a new asset you are trying on the older version?
We released 2021.2.1 to hopefully help with these issues.  It creates new AppData folders to hold assets for each version.  So if you open a older sim and they use 2021.1.1 assets, only that folder will be used.  I think the issue running the latest is that the api ids for the assets are not updated or wrong.  Can you replace the api ids with ids from the webui or new assets you have uploaded and see if that helps?

marcusvinicius178, 2021-07-30:
Sorry I did not understand the idea.....is there an easy way to clean all the old assets and update to have permission to use the new assets of the new release:

Take a look please hahahahahaha:

[image]

It is trying to dowload the plugins, maps for minutes...but never does... this is SVL 2021.1 

The new ones do not work because of what you have said I believe. I do not know ho to update, sorry.

EricBoiseLGSVL, 2021-07-30:
hmm, it shouldn't take long to download the keyboard sensor.  Do you have errors in the player.log? 2021.2.1 should create a new folder so there shouldn't be any issues running besides the template issue you reported.  My question is can you run an APIOnly template that runs python locally, no runner.
- check out 2021.2.1 tag make sure all other repos are on this tag
- delete all AppData folders for simulator
- run loader scene
- link cluster
- delete all assets in your library on webui.
- add map/vehicle/keyboard sensor to library that is labeled compatible with 2021.2.1
- create random sim with these assets and run
- create APIOnly simulation and edit a quick script to use map id and vehicle sensor config id,
- run to see if that works.
- If this works, then use API only not python api template

I think this is getting worse from jumping versions.  2021.1 saves to the default place and so does 2021.2 so you could be overwriting assets.

marcusvinicius178, 2021-07-30:
> hmm, it shouldn't take long to download the keyboard sensor. Do you have errors in the player.log? 2021.2.1 should create a new folder so there shouldn't be any issues running besides the template [...]

Unfortunately not. It was possible. But not anymore take a look:

[image]

> * check out 2021.2.1 tag make sure all other repos are on this tag
> * delete all AppData folders for simulator
> * run loader scene
> * link cluster
> * delete all assets in your library on webui.
> * add [...]

marcusvinicius178, 2021-07-30:
> Unfortunately not. It was possible. But not anymore take a look:
>
> [image]
I am using Apollo. You mean git checkout 2021.2.1 inside this folder?

> > * delete all AppData folders for simulator
Where is this folder located? Locally?

> > * run loader scene
What is this?

> > * link cluster
I have already unliked and linked many times
> > * delete all assets in your library on webui.

I have deleted the ones I used online, and then added again...no effect
> > * add map/vehicle/keyboard sensor to library that is labeled compatible with 2021.2.1
I jsut added automatically from the webstore in SVL website, just clicking...after remove, I clicked and added again

> > * create random sim with these assets and run
Yes I did just with a keyboard, random traffic....later with PythonAPi but any type it always starts dowloading and never ends to dowload..
> > * create APIOnly simulation and edit a quick script to use map id and vehicle sensor config id,
I did it
> > * run to see if that works.
> > * If this works, then use API only not python api template
> > 
> > I think this is getting worse from jumping versions. 2021.1 saves to the default place and so does 2021.2 so you could be overwriting assets.

YEs but is not anyway to delete everything from SVL online and locally to restart from scratch? I believe the issue is online..because it is dowloading but never finishes to dowload.

Thanks for your effort @EricBoiseLGSVL I really appreciate!

marcusvinicius178, 2021-07-30:
The error I get:

[image]

EricBoiseLGSVL, 2021-07-30:
```You mean git checkout 2021.2.1 inside this folder?```
simulator.  don't trouble shoot apollo at this point. we need to reduce vars ;)

```Where is this folder located? Locally?``` it is in AppData folder see [here](https://docs.unity3d.com/ScriptReference/Application-persistentDataPath.html).  If you haven't been deleting this then all old assets have been loading.  Be sure to delete this for all versions.

Try the app data delete, and relink and try again.  It seems if you can't connect you have a network error locally or server issues nearby. where are you connecting from?

I would also just make a second account to test if your backend data is messed up.  I'll ping @thisisdavidfrank and @eagafonovlge  to take a look at that

marcusvinicius178, 2021-07-30:
> `You mean git checkout 2021.2.1 inside this folder?`
> simulator. don't trouble shoot apollo at this point. we need to reduce vars ;)
Sorry in each folder mus I git checkout? I have already removed the folders inside the ~/.config/unity3D with rm -rf command line take a look:

[image]

> `Where is this folder located? Locally?` it is in AppData folder see [here](https://docs.unity3d.com/ScriptReference/Application-persistentDataPath.html). If you haven't been deleting this then all [...]
I am from south of Brazil
> I would also just make a second account to test if your backend data is messed up. I'll ping @thisisdavidfrank and @eagafonovlge to take a look at that
> 
> No worries, I apologize for the difficulties.
Thanks very much. This time it is really being hard to solve...I have removed everythin possible as I know , dowloaded the svl simulator again...but nothing helps.

marcusvinicius178, 2021-07-30:
This is my Player log from the folder ~/.config/unity3d/LGEletronics/SVLSimulator

```
Mono path[0] = '/home/autoware-auto-ros1/Área de Trabalho/svlsimulator-linux64-2021.2.1/simulator_Data/Managed'
Mono config path = '/home/autoware-auto-ros1/Área de Trabalho/svlsimulator-linux64-2021.2.1/simulator_Data/MonoBleedingEdge/etc'
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
Display 0 '0': 1920x1080 (primary device).
Display 1 'SyncMaster 19"': 1360x768 (secondary device).
Loading player data from /home/autoware-auto-ros1/Área de Trabalho/svlsimulator-linux64-2021.2.1/simulator_Data/data.unity3d
Unable to load player prefs
Desktop is 1920 x 1080 @ 60 Hz
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
Vulkan detection: 2
Initialize engine version: 2020.3.3f1 (76626098c1c4)
[Subsystems] Discovering subsystems at path /home/autoware-auto-ros1/Área de Trabalho/svlsimulator-linux64-2021.2.1/simulator_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1
[Vulkan init] extensions: count=17
[Vulkan init] extensions: name=VK_KHR_device_group_creation, enabled=0
[Vulkan init] extensions: name=VK_KHR_display, enabled=1
[Vulkan init] extensions: name=VK_KHR_external_fence_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_memory_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_external_semaphore_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_display_properties2, enabled=0
[Vulkan init] extensions: name=VK_KHR_get_physical_device_properties2, enabled=1
[Vulkan init] extensions: name=VK_KHR_get_surface_capabilities2, enabled=0
[Vulkan init] extensions: name=VK_KHR_surface, enabled=1
[Vulkan init] extensions: name=VK_KHR_surface_protected_capabilities, enabled=0
[Vulkan init] extensions: name=VK_KHR_xcb_surface, enabled=0
[Vulkan init] extensions: name=VK_KHR_xlib_surface, enabled=1
[Vulkan init] extensions: name=VK_EXT_acquire_xlib_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=0
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
[Vulkan init] extensions: name=VK_EXT_direct_mode_display, enabled=0
[Vulkan init] extensions: name=VK_EXT_display_surface_counter, enabled=0
[Vulkan init] SelectPhysicalDevice requestedDeviceIndex=-1 xrDevice=(nil)
[Vulkan init] Physical Device 0x1470540 [0]: "NVIDIA GeForce GTX 1050 Ti" deviceType=2 vendorID=10de deviceID=1c8c
[Vulkan init] Selected physical device 0x1470540
[Vulkan init] Graphics queue count=1
[Vulkan init] extensions: count=118
[Vulkan init] extensions: name=VK_KHR_16bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_8bit_storage, enabled=0
[Vulkan init] extensions: name=VK_KHR_bind_memory2, enabled=0
[Vulkan init] extensions: name=VK_KHR_buffer_device_address, enabled=0
[Vulkan init] extensions: name=VK_KHR_copy_commands2, enabled=0
[Vulkan init] extensions: name=VK_KHR_create_renderpass2, enabled=1
[Vulkan init] extensions: name=VK_KHR_dedicated_allocation, enabled=1
[Vulkan init] extensions: name=VK_KHR_depth_stencil_resolve, enabled=1
[Vulkan init] extensions: name=VK_KHR_descriptor_update_template, enabled=1
[Vulkan init] extensions: name=VK_KHR_device_group, enabled=0
[... 75 lines cut ...]
And here you can check there is any sensor, environment, plugin availbale...the folder are empty

[image]

At last but not the less important...After remove the python 3.8 I had HUGE problems, to run the simulator and docker again....
I needed to reinstall my Nvidia driver, the Cuda Toolkit, restart docker, removed the mesa-vulkan-drivers, reinstalled the libvulkan1 and vulkan-tools, etc..Then the SVL start at least to open without close fast....and the apollo was built on docker...
**BUT I have done some other MISTAKE in the process...I remember to fix other issue I have removed a PPA repo to have dowload access. But I do not remember which of them.. 
May this be the issue, what do you think?**

I just think because I have read this was the solution found by this guy: https://github.com/lgsvl/simulator/issues/753
But he did not said which PPA command is...

I tried the ones below found here: https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa

sudo add-apt-repository ppa:graphics-drivers/ppa
sudo apt-get update

I have also removed the docker image:

`docker image rm lgsvl/simulator-scenarios-runner:simulator-build__2021.1 
`

But still not success

I still have the same error...a loop in dowloading the assets...

EricBoiseLGSVL, 2021-07-30:
Sorry I am not familiar with PPA.  @lemketron ? I do know that sudo causes issues and shouldn't be used.  I think there are a few issues about it. Sounds like something is preventing connection and download.

lemketron, 2021-07-30:
PPAs are often used for installing pre-release software and that isn't the case for SVL Simulator.

I don't believe SVL Simulator requires (or relies on) any PPAs at this point.

Just follow the [installation instructions](https://www.svlsimulator.com/docs/installation-guide/installing-simulator/) including installing Nvidia drivers and Docker (neither of which mention any PPA repositories).

marcusvinicius178, 2021-07-30:
I understand I have followed all these installation instructions. So, what might be the problem of assets download? I have already cleaned all the system and restarted. I cannot imagine other factor than Network. But I am just a student... don't know what is going on

marcusvinicius178, 2021-07-30:
But now sometimes my simulation stops apparently without reasons:

And say 

[image]

Also this image when turns offline

[image]

marcusvinicius178, 2021-07-30:
When the error above happens. I need to stop the Dreamview:

`./scripts/bootstrap_lgsvl.sh stop`

and stop the bridge

close the simulator, open again and restart everything...
In addition the 2021.2.1 SVL release put the car in the wrong position, when I trigger the localization module:

[image]

In previous versions, the car always was put correctly in Dreamview when I activated the localization module...

Do you imagine what is generating this localization error?

marcusvinicius178, 2021-09-06:
Hi @EricBoiseLGSVL sorry to reopen this issue. It is just because I am trying to standardize the simulator release I am using for all the machines I use...

And in my notebook I have redone the steps we discussed above:

1 - remove the docker image **scenarios-runner-2021.1** and the other ones
2- removed the LG folders inside **.config/unity3D**
3- Unlink cluster, clear cache, delete and dowload again all the assets (maps and vehicles)

Then I have removed and dowloaded again the PythonAPI folder and installed it. Because it seems the scenarios runner are in someway linked to it. Then I thought when I runned the installation command again inside this folder I would have the scenarios runner from the new SVL release. But I got these messages:

[image]

And now if I try any version of SVL I get the same error:

If I try with the new release (2021.2.1) I wish I get:

[image]

And If I try with the old one I was using (2021.1), the same error:

[image]

I cannot run any simulator release. Do you know what I could try to start a "fresh system" regarding simulator/linux without need to reinstall ubuntu? It seems some old " PythonAPI contents" related to the old SVL release remained on linux folders....and I am not sure if I can delete these folders the terminal displayed. Some idea?

Thanks

marcusvinicius178, 2021-09-06:
As @lemketron said:
I installed Apollo 6 today with the LGSVL tutorial: https://www.svlsimulator.com/docs/system-under-test/apollo-master-instructions/

git clone https://github.com/ApolloAuto/apollo

I am not sure why it dowloaded the docker image for the 2021.1 runner-scenarios

I believe you're seeing leftover docker images cached on your machine from having previously used the scenario runner in SVL Simulator. This is not related to apollo and should not affect it.

It sounds like you are changing simulator and/or apollo versions around, and it's not entirely clear which kind of simulation you are trying to run. Which runtime template are you trying to use? Are you using Python API (which uses a built-in python scripts) or API-only mode? If you are trying to run a local python script with the simulator, then you'll need to use API-only mode in your simulation. This cannot be changed once you have created a simulation but you should be able to make a new API-only simulation. At that point the errors related to the runner container should go away.

But If I wish to use the Python-API template how can I scape this error and run the simulation?

marcusvinicius178, 2021-10-01:
hI @EricBoiseLGSVL I have an URGENT request please:

[image]

I have created a new e-mail account: 
marcusvinilgsvl@gmail.com

To manage my final master thesis simulations...
And my limit to create clusters was reached could you please enable it as soon as possible? 

Other issue faced again is:

I need to use the LGSVL 2021.1 release instead the 2021.2.1 release, because I checked on my Notebook machine that the newer release (2021.2.1) fails to localizate the car in SanFrancisco map, take a look:

[image]

While the old release (2021.1) does not:

[image]

However I needed to run the old release in those rented cloud machine I told you, because on my notebook the planning module does not work (due low CPU capability).

And the **issue ** is that when I tried to install the LGSVL old release and run the simulations I am facing all those problems we discussed above....I have done everything you said:

- Removed All Assets
- Removed the LGEletronics folder inside .config/unity3D folder
- Cleared the cache
- -Unlinked and later linked the simulator with web Browser
- Created a new cluster
- Pruned dangle images

However the simulator goes offline when I run the simulations with PythonAPi Template. I cannot use the PythonOnly template in this virtual machine I  have rented, because I had lot of issues to install the PythonApi due sudo permissions not available or/and others....

Could you help me please?

EricBoiseLGSVL, 2021-10-01:
I contacted backend to increase your clusters.  We are trying to push out the 2021.3 release right now which will fix all these issues.  Hang in there, we got you.

marcusvinicius178, 2021-10-01:
Hi Thanks a Lot!!! I cehcked the website was in maintainance.

I hope this new release be amazing as everything LG does!!! 

Of so I will be waiting...

I have done nice simulations with Autoware.AI and Apollo....I could use almost all the planners from the 2ADS....and I am doing nice plots to compare the planners and the stacks regarding the modules (localization, perception,planning,control,etc)

Unfortunately just AutowareAuto was not used, because of those bridge issue. **Would you know if AutowareAuto is back to work with SVL???** I would like to use their AVP planner with SVL....
****

@EricBoiseLGSVL  would you know how could I produce a **Point Cloud Video** from these simulations, as the **bird-view** nice feature that the video-recording sensor plugin does?

Is this just for "Premium users" Would you suggest me something to have PCD video while the car travels?

Thanks in advance

lemketron, 2021-10-03:
> Unfortunately just AutowareAuto was not used, because of those bridge issue. **Would you know if AutowareAuto is back to work with SVL???** I would like to use their AVP planner with SVL....

Please start a new github issue for this, but I am definitely interested in hearing more about your bridge issue with Autoware.Auto. We are actively testing Autoware.Auto with SVL Simulator and expect that 2021.3 should resolve many outstanding issues for Autoware.Auto users -- though I was not aware of any bridge issues.

Please be sure that you are using ROS2 with Autoware.Auto and not ROS (1) as with Autoware.AI. And again, if this continues to be a problem for you with 2021.3 (which you should be able to try next week) then please file a new github issue with the details (and logs). Thanks!

marcusvinicius178, 2021-10-03:
Hi Steve I am referring about these issues https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/977

I was working normally with AUTOWARE.Auto and have done simulations in Autonomous Stuff scenario with the AVP (Lanelet2) path planner. However LG discontinued the 2020.06 simulator and then the new releases were not working with Autoware.Auto as it is related in the link above....

I am happy that the new release will solve all the issues. I am just unhappy that my machine now cannot run the SVL 2021.2.1 or 2021.1 after I use them one after anothet with PythonApi. I don't understand how to fix this and be able to use more than 1 SVL version without bugs...
I am not sure now what should I do, if erase all Ubuntu and start from a clean state or what? Because deleting the LGELETRONICS folder inside the .config/ folder was done, the cache was erased, the plugins, vehicles and maps on web app was one by one all removed by me, I have unlinked the Cluster and then created new one....but anything I do is able to enable the PythonApi template back to work....☹️
Would you have some suggestion please? I would not like to erase Ubuntu from scratch

lemketron, 2021-10-03:
You should be able to try Autoware.Auto with 2021.3 very soon (within the next few days I hope)...  :-)

I'm sorry for the troubles you had trying to use Autoware.Auto with 2021.1 and with PythonAPI with other 2021 versions. Please try all of that again with 2021.3 and let us know if you discover any new or remaining issues.  Thanks!

closed 2022-03-13 by marcusvinicius178