lgsvl/simulator #1647: Cannot sometimes stop the sim

krtt, 2021-07-29:
Sometimes when pressing 'Stop Simulation' in the browser UI, the state changes to 'Stopping' and then just hangs there, without actually fully stopping. This time, it happened after running ca 6 times totally ok, then suddenly got stuck and only restarting the sim program helped. Player.log attached.
[Player.log](https://github.com/lgsvl/simulator/files/6901788/Player.log)

[image]

Ubuntu 18.04
Sim v2021.2 (API Only mode)
Apollo 6.0

marcusvinicius178, 2021-07-29:
This happens to me sometimes, then I always stop the simulation using the simulator tab: Stop...in simulator window program, not in SVL website...it always works and close the simulation

EricBoiseLGSVL, 2021-07-29:
Yes, this is what I do as well.  We do have @heeen looking into a possible solution.  Thanks for posting.