lgsvl/simulator #1649: Default constructor not found!

duongtran11, 2021-07-31:
I run the build for a Ros bridge, got no error, then I upload the bridge to webUI, but it didn't work as expected. I think there's something wrong with the build process. And there is a suspicious warning (in the below image). Is it something I should worry about?
[image]

heeen, 2021-08-02:
Are you trying to build ROS Bridge as a plugin/assetbundle? I think your ROS bridge type is conflicting with the one that is part of the simulator codebase in 2021.1.1

heeen, 2021-08-02:
FYI ROS/ROS2/CyberRT Bridges as plugins is an upcoming feature (https://github.com/lgsvl/simulator/commit/d090e920425fc14cd00763da6f479b1db39b9062)

EricBoiseLGSVL, 2021-08-02:
Yes, @heeen is right, custom bridges need 2021.2.1 release tag.

duongtran11, 2021-08-05:
> Are you trying to build ROS Bridge as a plugin/assetbundle? I think your ROS bridge type is conflicting with the one that is part of the simulator codebase in 2021.1.1

You're right. I had to build the whole simulator to make it work. but then I had to build a Ros bridge myself cause pre-build Ros bridge on the webUI doesn't support dev version of simulator.
Can you change that?

EricBoiseLGSVL, 2021-08-06:
We have support for development in 2021.2.1 version of simulator.

closed 2021-08-16 by duongtran11