lgsvl/simulator #1653: Using sim in Unity with Intel HD graphics 520 card

brookebroccoli, 2021-08-03:
Hey there,

I'm not 100% sure on how exactly the simulator works so bear with me if I make any mistakes in my logic. So from what I understand the simulator is its own standalone platform, and using Unity enables developers to alter aspects of the simulator as they like?

I am wanting to know if a graphics card is necessary to **build and to run** the simulator in Unity? When developing the game in Unity, is a graphics card necessary? I'm using a surface book 1 (with ubuntu 18.04 as my os) which has an Intel HD graphics 520 card but I'm wondering if I can still work with the software for the sim. 

I have been trying to build the simulator from source but haven't had much luck in doing so and I'm just hoping there's a way to work with the simulator in unity without needing a graphics card. When I build just the simulator (with no assets) I don't get errors but I get a bunch of warnings - I have pasted the Player.log file below which shows that Vulkan doesn't support the graphics card. When I do include an asset as per the instructions [here](https://www.svlsimulator.com/docs/installation-guide/build-instructions/#building-assets), for example [cube town](https://github.com/lgsvl/CubeTown) and git clone the repo and try build the environment I get a bunch of errors. I tried to see if vulkan is supported on the Intel HD Graphics 520 card but I haven't found too much. 

I know that the lack of a vulkan supported graphics card is probably the issue but I just had hope that maybe there is a work around. Sorry for the super long response, would appreciate any help!

**Player.log**

Mono path[0] = '/home/brooke/Documents/simulator/TheBuild/simulator_Data/Managed'
Mono config path = '/home/brooke/Documents/simulator/TheBuild/simulator_Data/MonoBleedingEdge/etc'
PlayerConnection initialized from /home/brooke/Documents/simulator/TheBuild/simulator_Data (debug = 0)
PlayerConnection initialized network socket : 0.0.0.0 55112
Multi-casting "[IP] 192.168.0.22 [Port] 55112 [Flags] 2 [Guid] 1532356099 [EditorId] 1711486093 [Version] 1048832 [Id] LinuxPlayer(192.168.0.22) [Debug] 0 [PackageName] LinuxPlayer [ProjectName] SVLSimulator" to [225.0.0.222:54997]...
Started listening to [0.0.0.0:55112]
Preloaded 'UnityFbxSdkNative.so'
Preloaded 'liblaszip.so'
Preloaded 'libpng.so'
Preloaded 'libsqlite3.so'
Preloaded 'libturbojpeg.so'
PlayerConnection already initialized - listening to [0.0.0.0:55112]
Display 0 'VVX14P048M00 14"': 3000x2000 (primary device).
Loading player data from /home/brooke/Documents/simulator/TheBuild/simulator_Data/data.unity3d
Desktop is 3000 x 2000 @ 60 Hz
[Vulkan init] extensions: count=2
[Vulkan init] extensions: name=VK_EXT_debug_report, enabled=1
[Vulkan init] extensions: name=VK_EXT_debug_utils, enabled=0
Vulkan detection: 0
No supported renderers found, exiting

lemketron, 2021-08-03:
> I know that the lack of a vulkan supported graphics card is probably the issue but I just had hope that maybe there is a work around.

Yes, this is the problem. 

On Windows the Simulator may run on a non-Nvidia GPU (like some integrated GPUs) but the performance probably won't be acceptable and some features might not work correctly. 

On Linux, the Simulator needs a GPU with Vulkan support (to properly support Unity's High Definition Rendering Pipeline or HDRP). Currently we recommend using a Nvidia GTX-1060 (Pascal) or RTX-2060 (Turing) or 3060 (Ampere) or equivalent or better (preferably with 8GB of GPU memory as 4GB may not be enough for some of the Simulator's 3D environments).

To test that your GPU and drivers are installed and working properly for Vulkan, try (installing and) running `vkinfo` and `vkcube`.

brookebroccoli, 2021-08-03:
Damn that's a shame but anyway, thanks for the speedy reply!

heeen, 2021-08-03:
I was able to barely run it on linux on a Thinkpad with a nvidia MX150 but I had to really scrape for free gpu memory by reducing screen resolution and closing other windows. It ran cubetown to be able to do some smaller tests though.

closed 2021-09-22 by EricBoiseLGSVL