lgsvl/simulator #1655: Sensor fidelity

tajinder96, 2021-08-03:
Hi, 

Had a quick question about where I could read more about the fidelity of the sensors (camera, radar, lidar). 

I understand that weather conditions only influence the camera sensor. I would like to learn about the other factors that influence the camera, lidar, or radar sensors.

Thanks!

EricBoiseLGSVL, 2021-08-05:
Weather effects and sensor effects are only for color camera sensors.  These other sensors listed are not effected by weather but the code base is open source for these sensors and you can add whatever effects you need.

tajinder96, 2021-08-06:
Thanks, Eric. May I know in general what level of fidelity is available for the sensors? For example, in terms of how realistic they are w.r.t shape, material properties.

EricBoiseLGSVL, 2021-08-06:
Well we have about 45+ sensors we have created internally, it's difficult to break down each one.  They range from realistic to simplified to give users a starting point or to give the AD stacks were work with exactly what they need to operate.  In the spectrum with have the VelodyneLidarSensor that we modeled after the exact specs of the hardware and works with the real driver.  Then we have our lidar sensor that is one of the most efficient and accurate model for simulation using cube maps. Camera sensors that you can edit to simulate many different camera types, Then we have a radar sensor that is only trigger volumes that return colliders.  This is why we open source most of our sensors so users can use our work as a starting point to create specific sensors they need.

tajinder96, 2021-08-07:
Thanks a lot for this information :)

closed 2021-08-07 by tajinder96