lgsvl/simulator #1657: Simulator cannot go online

brookebroccoli, 2021-08-04:
Hey guys,

I ended up borrowing a friends laptop which has a NVIDIA GeForce GTX 1660 Ti graphics card. Its a surface book 3 and I'm running ubuntu 18.04, Unity 2020.3.3f1. I've installed the nvidia drivers (470.57.02) and I'm building the simulator from source as a development build, but when I open the simulator.exe file after building, I cannot go online and there is an error in the bottom left saying argument out of range (see screenshot below of GUI and player log is attached). One of the drop down options in the simulator GUI is to clear the cache, which I tried but that didn't help anything.

I saw on [this](https://githubmemory.com/repo/lgsvl/simulator/issues/1609) post that a usb internet dongle is necessary to get the sim to work, but I haven't seen that written anywhere on the svlsim website so wasn't sure if that is true or not, or if there is something else that is causing the the problem.  

Would really appreciate any help with this!

[Player.log](https://github.com/lgsvl/simulator/files/6928408/Player.log)

[image]

EricBoiseLGSVL, 2021-08-05:
Welcome to svl simulator.
- gtx 1660ti is below minimum spec so be sure not to run your AD stack on the machine at the same time. 4K screen will hurt also :)
- go to application and look for config.yml file.  If not there, copy from the repo root directory
- if you have ```cloud_url: https://wise.svlsimulator.com/``` remove the trailing ```/``` and no quotes on url

If that didn't work
- did you install windows mono support before building the application?
- were there any errors in the console when you built the application from Simulator -> Build menu?
- what release tag version of simulator are you building from?

FYI that post is not saying that a dongle is needed but is not supported.  We expect users to connected to a hard wired port and dongles/wifi is not supported.  You can try and please post how if you are successful.

Hope this helps.

JJ-GYZ, 2021-09-08:
[image]
I also encountered the problem of not being able to connect to the cloud. It doesn't work to click the button repeatedly. I think the graphics card version may be too low. It's stuck in the first step like this.

EricBoiseLGSVL, 2021-09-08:
What is your default browser?  Do you have any software blocking links to the browser? When you click Link to Cloud, it just opens the browser to svlsimulator.  Please try to open the URL before (also be sure you are signed in) you click it and see if that helps.