lgsvl/simulator #1658: Adding a vehicle on apollo with more lidar

Sal-96, 2021-08-04:
Hi, I am using the lgsvl simulator (built by source 2021.1.1) with the latest version of apollo, I have inserted a custom vehicle and then added sensors via web ui. I have followed the documentation and I have modified the file vehicle_param.pb.txt, I wanted to understand how to add for example on apollo more lidar and ultrasound sensors. I need them for autonomous driving.

EricBoiseLGSVL, 2021-08-05:
You can easily add more lidars and ultrasonic sensors to the vehicle in svl simulator but it is another thing to set up in Apollo.  You will need to update the vehicle config and set topics.  Anything else @lemketron ?

Sal-96, 2021-08-05:
There are some exemple to  set up in Apollo?