lgsvl/simulator #1661: Import Autoware Custom assets give error

RDaneelOlivav, 2021-08-04:
Hi,

I'm trying to import the following assets from the example of AutowareAuto tutorial in the latest version of SVL simulator, and its giving the following error:

[image]

These are the assets:

Vehicle: https://assets.lgsvlsimulator.com/ea5e32fe566065c6d1bbf1f0728d6654c94e375d/vehicle_AWFLexus2016RXHybrid

Maps ENv Asset: https://assets.lgsvlsimulator.com/ec057870762b5a967a451c93444b67d0b64e9656/environment_AutonomouStuff

I dont really understand whats wrong here, any ideas? I tried this a month ago and worked but now the web client has changed and it doesnt work at all. Any ideas or suggestions of what it could be? 

Thanks in advancd

EricBoiseLGSVL, 2021-08-05:
- In the webUI for the latest release 2021.2.1 we have uploaded all assets to maps/vehicles and you can add to your library. Older assets are not compatible with latest releases. Where did you get these assets? The documentation says this [here](https://www.svlsimulator.com/docs/system-under-test/autoware-auto-instructions/#run-simulator-alongside-autoware-auto).
- If you have local assets that are old I recommend deleting in the ClearCache menu before running new versions. See [here](https://www.svlsimulator.com/docs/user-interface/simulator-main-menu/).

RDaneelOlivav, 2021-08-05:
Hi,

Is because I was following the instructions in Autoware.Auto documentation: https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html

But today I followed the instructions of the page you gave me: https://www.svlsimulator.com/docs/system-under-test/autoware-auto-instructions/#run-simulator-alongside-autoware-auto

After some issues, I managed to have the simulator launched, and the ade with all the needed packages installed, as far as I can see.

I've created a video to show the main launching and the issues that there are so that is easier to understand and explain:

[VIDEO SHOWING THE ISSUE](https://youtu.be/yjvJkF3S01s)

But some issues arrived:
1) The simulation DOESN'T publish the TFs so nothing is visualized
2) Even if this would work, as far as I know, the Autoware.Auto demos are based on a map that is NOT in the library ( San Jose ). Am I wrong? Is there an Autoware navigation demo that I'm missing?

So any suggestion or idea that might help with this will be welcomed because I want to use Autoware.Auto and SVL simulator but its becoming more and more arduous to just have it running. And seems other people are having [similar issues](https://answers.ros.org/question/381248/how-to-set-up-the-autowareauto-with-the-svl-simulator/).

EricBoiseLGSVL, 2021-08-05:
Sorry you are having issues.  I'm not sure why there are different docs on AutowareFoundation's site.  I'll ping people internally to see if they know what the issue is.

lemketron, 2021-08-05:
The [documentation on the Autoware site](https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html) still refer to the older 2020.06 version of LGSVL Simulator.

The [documentation on the SVL Simulator site](https://www.svlsimulator.com/docs/system-under-test/autoware-auto-instructions/) is for the new (with all new web-based UI and different asset bundles, etc.) 2021 versions of SVL Simulator.

We will work with the Autoware folks to get their docs updated. Bottom line is that although old, the Autoware docs work with the older simulator, and if you're using SVL Simulator 2021, you should (for now) refer to our documentation since it is updated for each new version that we release.

RDaneelOlivav, 2021-08-06:
Thanks for clearing this up, I now understand why this is not working at least part of it.

Although the video I'm showing is for the simulation tutorial you show in the 2021 version, and the TFs are not published, which doesn't allow then for the sensor messages to be used. 

Do you know any reason why this could happen?
Do we have to set the TF publication somewhere in the vehicle definition?

hadiTab, 2021-08-06:
@RDaneelOlivav The simulator does not broadcast tfs. There is normally a ROS node responsible for that. The simulator just allows you to configure the frame names in the messages so that they match would be expected.

It seems like our documentation is missing a step. You will either need to manually broadcast static transforms or use the `robot_state_publisher` package. 

I believe you should be able to get `robot_state_publisher` up and running using [this](https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/blob/master/src/launch/autoware_auto_launch/launch/autoware_auto_simulator.launch.py) launch file in the AutowareAuto repository. Make sure that the `lgsvl_bridge` is running.

RDaneelOlivav, 2021-08-09:
Hi,

Thanks @hadiTab for pointing to the correct launch ;).

It worked with the newest version: [VIDEO](https://youtu.be/5V1PqvJ5P5M)
Also with the old version did exactly the same: [OLD SVL VIDEO](https://youtu.be/KfJp36typaI)

Now returning to the original question, the assets are given [here:](https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/lgsvl.html)

Happens this error: [VIDEO ERROR Uploading Assets](https://youtu.be/QPrZeSA_6To)

[Vehicle: ](https://assets.lgsvlsimulator.com/ea5e32fe566065c6d1bbf1f0728d6654c94e375d/vehicle_AWFLexus2016RXHybrid)
[Maps ENv Asset:](https://assets.lgsvlsimulator.com/ec057870762b5a967a451c93444b67d0b64e9656/environment_AutonomouStuff) 

**Give error when trying to upload then into the newest version.**

Is there somewhere where these assets are upgraded to the newest version of SVL? Or somehow to upgrade them for the newest version?

Any ideas?

hadiTab, 2021-08-10:
@RDaneelOlivav the upgraded versions are available by default in the store.

This is the Lexus: https://svlsim.com/vehicles/profile/b22952f6-f82b-4eeb-ac28-2067e88569a1
This is the AutonomouStuff map: https://svlsim.com/maps/profile/2aae5d39-a11c-4516-87c4-cdc9ca784551

You just need to add them to your library and then use them in a simulation. Look at the getting started docs if you haven't already to get a better idea of how to run simulations. For example: https://www.svlsimulator.com/docs/getting-started/getting-started/

The assets you are trying to run are older versions that are not supported anymore.