lgsvl/simulator #1662: Simulator and Rviz2 cannot run alongside on new Release?

marcusvinicius178, 2021-08-04:
Hi @EricBoiseLGSVL could you please assist me on this issue?

I am running the SVL 2021.1.1 release with AutowareAuto bridge on BorregasAve map. The bridge connection is working and ROS2 is subscribing the data from the simulator topics. However when I try to open rviz2 in other terminal, it does not get opened and raises a dbus_connection issue. It says that "the address is already in use" take a look:

[image]

This was a launch file tried...However If I try to open rviz2:

[image]

A similar error

EricBoiseLGSVL, 2021-08-05:
I would first check to see if any other applications are using your ports but I'm not sure past that. @hadiTab or @cyberphysicalrobotics ?

marcusvinicius178, 2021-08-05:
Sorry I have already solved here: https://gitlab.com/ApexAI/ade-lgsvl/-/issues/8

closed 2021-08-05 by marcusvinicius178