lgsvl/simulator #1663: Get PCD maps for SVL maps

marcusvinicius178, 2021-08-05:
Hi @EricBoiseLGSVL  would you know how I get maps to use with AutowareAuto:

https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/1255

EricBoiseLGSVL, 2021-08-05:
We used to have one in simulator when the lidar was is source but we had to disable it.  I'll see if we can enable again in the next release.  If you want, you can look at PointCloudExport,cs in the meantime. It basically spawns a lidar object and follows the map lanes in a scene.  You can also drive a vehicle around the map and capture the point cloud.

marcusvinicius178, 2021-08-06:
Hi @EricBoiseLGSVL nice idea (you mean drive the car along the map to create the PCD as in Navigaton Stack ROS1 functionality right? There I drived the car to create a map for SLAM...but this consumes a lot of time and effort. 

For this reason I had the idea to get the .pcd files already available on repo: https://github.com/lgsvl/autoware-data/blob/master/BorregasAve/data/map/pointcloud_map/BorregasAve.pcd

Then I have modified and created new launch files from ones given in the AVP autonomous stuff demo: https://autowarefoundation.gitlab.io/autoware.auto/AutowareAuto/avpdemo.html

In order to target into the new PCD file (BorregasAve instead AutonomousStuff pcd), so I replaced the:

avp_sim.launch.py for borregas_avp_sim.launch.py

[image]

And then inside this file I have called for the borregas_map_publisher_sim_param.yaml (which I created) instead the map_publisher_sim_param.yaml. The difference is taht this new yaml file calls the PCD of BorregasAve (provided in AutowareAI repo) instead the standard AutonomousStuff PCD. 

[image]

Finally I have build the workspace:

```
$cd AutowareAuto
$colcon build
```

And then I have launched the new launch file which contains the BorregasAve  PCD. However AutowareAuto gets this launch file from the /opt folder, which is read-only by default. I tried to change the permissions to copy the launch file inside this folder, with the hope it could be further launched...however I had no success.

[image]

In this way I would like to understand how to use different maps on AutowareAuto to visualize on rviz2?

In addition I am not sure if just exchange the PCD file will work, because there is another file target called autonomousstuff_parking_lot_lgsvl.yaml that has some parameters such as longitude/latitude and other ones that I am not sure how to configure for another map, such as BorregasAVe:

[image]

Do you have some hint on this regard? Thanks in advance!

EricBoiseLGSVL, 2021-08-06:
Sorry I am not sure how to force this in Rviz2.  Maybe someone else has done this.  I would try to post in autoware forums.  Still, we have had issue with users running examples from autoware since it is very out of date. We are looking into trying to get those updated.

marcusvinicius178, 2021-08-06:
UPDATE:

Actually I have found one error (from my part). Actually there was an update of the avp tutorial:

And I should source other file (install/setup.bash instead the /opt/AutowareAuto/setup.bash)

[image]

Then my launch file with the BorregasAVe PCD file (get from the AutowareAI repo) was targeted!!!

However now I still face errors in rviz2 (with Transforms and others):

[image]

And there is just an explanation for that! The PCD file provided by AutowareAuto (AutonomousStuff) and the configuration file, are different from those ones provided by AutowareAI, and therefore I cannot use it....
This is a big problem for me because I have no idea how and where to get PCD and configuration files compatible with AutowareAuto for the other LGSVL maps...(Borregas, SanFrancisco).

Thanks for the answers @EricBoiseLGSVL  I will try the AutowareAuto forums!
But if you have some idea in the future where could I get these PCD and config fiels for the other maps I will be very thankful. I believe these files are generated in partnership AutowareAuto-LG or not? Just the AutonomousStuff? Just AutowareAuto is responsible to generate and provide these files?

EricBoiseLGSVL, 2021-08-06:
Excellent and yes we have the team looking at getting our point cloud feature working again. Not sure on how these were setup but once we get the manual way working we can look into adding it to the pipeline.

closed 2021-08-06 by marcusvinicius178