lgsvl/simulator #1665: How to obtain fisheye parameters according to the original factory information table

selfsimulator, 2021-08-05:
I am fitting the distortion formula through the original factory information table, and the distortion formula of K value can be well fitted. But it is fitted by the formula of the unified projection model ξ The value is particularly large, and the curve error is particularly large! So I'd like to ask lgsvl how to obtain the fisheye configuration parameters - Distorted parameters and ξ ?

[image]

[image]

[image]

EricBoiseLGSVL, 2021-08-10:
What original factory information table? What are you trying to do exactly? What isn't answered in #1498? After looking at your linked repo, it seems you have copied the code and removed the copyright headers.  Are you aware that this is not allowed?  I'll add @kamkolak-lge to explain more on the fisheye parameters

selfsimulator, 2021-08-12:
> the copyright headers

Sorry, I didn't notice the copyright title. I have added it again and deleted the GitHub library. And my question is, how can we accurately determine the K and Xi when using this fisheye camera?

kamkolak-lge, 2021-08-13:
@selfsimulator
I'm not sure what you're trying to fit. Distortion for camera with fisheye enabled is done using unified projection model. Aside from the standard camera parameters like resolution and field of view, you only need to define offset for center of projection (ξ, `Xi` in camera parameters). I'm not sure where the formula you used comes from, but the math behind this type of projection can be found, for example, [here](https://www.robots.ox.ac.uk/~cmei/articles/single_viewpoint_calib_mei_07.pdf).

If you look at how the [compute shader](https://github.com/lgsvl/simulator/blob/90c47ce664da6f993da5e7f463213e6f0c667272/Assets/Shaders/LensDistortion.compute#L68) implements it, you should be able to map it to equation (2) from linked article.

Please note that it's possible to use plumb bob distortion on top of unified projection model, so if your coefficients (`DistortionParameters`) are non-zero, you have to take this into account.

selfsimulator, 2021-08-16:
> @selfsimulator
> I'm not sure what you're trying to fit. Distortion for camera with fisheye enabled is done using unified projection model. Aside from the standard camera parameters like resolution [...]

I want to know how to calculate the standard camera parameters and offset for center of projection (ξ, `Xi` in camera parameters) ? Because I don't know what the appropriate size of these parameters is ? Are these parameter values based on real fisheye camera data? If not, the values of these parameters can be entered by the user at will and there is no real data support, then the fisheye camera is meaningless!
[image]
[image]

selfsimulator, 2021-08-16:
> @selfsimulator
> I'm not sure what you're trying to fit. Distortion for camera with fisheye enabled is done using unified projection model. Aside from the standard camera parameters like resolution [...]

And I want to know what this part of the code means? Is it part of a plumb bob model? I can't relate it to a plumb bob model.
[image]
[image]