lgsvl/simulator #1668: how to combine 5 point cloud files (map.pcd) or  HD map files (base_map.bin) with different map origin.

Addi2020, 2021-08-05:
I have 5 .pcd files of 5 different roads that are adjacent to each other. So how can I create on HD map (base_map.bin) using these 5 files , all having different map origin. 

What I tried?
1. I import 1 .pcd file then try to import another .pcd file in the same scene or project, but it replace the old .pcd file instead of having two files. Even If i import two files then how to adjust the map origin?
2. I import 5 .pcd files all together, but it overlaps with each other and I cannot see the actual road, also I cannot move these files.
3. I also tried to import two base_map.bin files in one scene, but it give an error. I thought I can create 5 different HD maps (base_map.bin) and then i can combine these files but it is not possible.

So how to deal with this issue?

EricBoiseLGSVL, 2021-08-06:
We do not support multiple pcd files at this time. This would be an extensive amount of work.  Why do you have 5 different files? I suggest making one map per file.  Why do you need to test an area that large?

Addi2020, 2021-08-07:
@EricBoiseLGSVL  Thanks. Its our company requirement to create huge map of a big city for testing full autonomous driving in large area of a city.

EricBoiseLGSVL, 2021-08-11:
I see. When you create any environment in a game engine, you run into limitations of how much memory can be used for a map.  This is why games have blocked off areas to load as a player moves through.  Think of GTAV and how on start, the game doesn't load the entire map into memory. I recommend you scope the scale of the environment and test what you need in each section of the map per environment.
If that can't work for your needs then you will need to load/save each section into a scene and then manually edit one scene to hold all the combined data.

Here is an example I did with CubeTown and BorregasAve.  I opened one scene and then dragged in the other scene.  Unity allows multi-scene editing.  Notice I created holder objects for the entire scene so you only need one object to drag in.  Then separate the annotation into the one annotation holder by dragging lane sections and intersections. Save the main scene and delete others before making the final bundle.
[image]

Addi2020, 2021-08-20:
@EricBoiseLGSVL Its good, I also tried to import three HD maps and it works fine. So it means that we can create small HD maps and then we can import HD maps to combine them all together to create big_map. 

[image]

closed 2021-09-06 by Addi2020