lgsvl/simulator #1669: 1:10 Car

digget, 2021-08-05:
Hey,
i have scaled the Jaguar2015XE to be an 1:10 scaled Car.

Driving with Keyboard Controller Video:
https://youtu.be/jUcgYwGnCmQ

Issues faced:
Front Wheel slipping hard -> Traction Problems

Driving with the F1TENTH Car is working so smooth.
Has anyone an advice how i could solve my traction issue ?
Also when i steer, the car gets slower.
I tested almost every value.

Model Settings:
[image]
[image]

WheelCollider Settings:
[image]

[image]

Car used: https://github.com/lgsvl/Jaguar2015XE
Using Unity Version 2020.3.3f1
With branch LGSVL Branch 2021.2

Thanks to everybody

ericSterner, 2021-08-05:
Hi Digget, unfortunately scaling down prefabs leads to a number of physics problems as you have discovered.  For best results we create the 1/10 scale vehicles/prefabs from scratch so physics settings remain 1:1.  Your best option would be to use the F1TENTH Car.

digget, 2021-08-05:
Hi @ericSterner,
i have the same problem with 1:1 1/10 Cars. Created the model within blender, so that everything is scaled 1:1

[image]

EricBoiseLGSVL, 2021-08-05:
We have an update that will let users set mass of the rigidbody and set their wheel colliders.  This will release soon.

digget, 2021-08-05:
@EricBoiseLGSVL What will be the difference to build the Car within Unity ? 
Like in this Tutorial explained: https://youtu.be/R_Z65kimCII

EricBoiseLGSVL, 2021-08-06:
The difference will be you can have a rigidbody and wheel colliders in the npc prefab and we use that data instead of the default NPC data for regular sized vehicles.  The reason your npcs don't work is that they are set to regular vehicle mass and wheel collider settings.  This is set in code.  I'll make a note to screen shot the f1tenth in the docs so everyone can see the settings for small vehicles.

digget, 2021-08-06:
@EricBoiseLGSVL i dont use npc vehicles. The Vehicle i want to controll is the problem. Every information about the f1tenth car you are able to share would be great ! Thank you

EricBoiseLGSVL, 2021-08-06:
Oh I see. Let me take some screen shots of the physics settings.  It should help.
[image]

[image]

@digget one thing I noticed is that your pivot for the main mesh might not be at the bottom.  See the the first image I posted.  I can't tell what gameobject you have selected.