lgsvl/simulator #1671: Lexus vehicle model

krtt, 2021-08-06:
Is it possible to get the Lexus2016RXHybrid vehicle model as you have provided in the Jaguar example [here](https://github.com/lgsvl/Jaguar2015XE/tree/master/Models)? The Lexus downloadable asset bundle is not suitable.

EricBoiseLGSVL, 2021-08-06:
Sorry, why is the Lexus asset bundle not suitable?

krtt, 2021-08-09:
The asset bundle holds these files:
[image]

I tried to find information to access and browse the vehicle model in this bundle. I found the Unity Asset Bundle Browser tool. But when I tried to open it with this tool, it does not show any additional files. How to work with this bundle?

Ubuntu 18.04
Unity 2020.3.3

EricBoiseLGSVL, 2021-08-10:
We do not provide source for the Lexus2016 model.  You cannot and should not try to extract it from the bundle.  This is a model purchased and we cannot make it open source.  If you need a specific model vehicle, there are many websites that will sell game ready assets.

krtt, 2021-08-11:
From where did you purchased it? So we can purchase it too from the same place. If possible.

ericSterner, 2021-08-11:
https://hum3d.com/3d-models/lexus-rx-hybrid-2016/

closed 2021-09-28 by krtt