lgsvl/simulator #1675: map is worng in apollo's dreamview

OYhhh2, 2021-08-10:
the instructions to add other maps in svl to the apollo (my verision is apollo 6.0)
[image]

However,the map seems wrong ,the car is not in the right possition in dreamview , the simulator has no problem, It seems that the map have a rotation ? i don't know what' s the problem. the map is SanFrancisco,and Shalun also havs this problem
[image]

Can you help me ?Thanks!

EricBoiseLGSVL, 2021-08-10:
Which version of simulator are you using? Where did you get these maps? Maps were rotated -90 degrees to ensure proper heading values in GNSS related topics.  You can read about it [here](https://www.svlsimulator.com/docs/getting-started/conventions/).

OYhhh2, 2021-08-11:
The version of simulator is 2021.2,and i get these map from the simulator's library,click the apollo50 so i can get the map file and follow the instructions to generate the map file that can use in apollo.
[image]

OYhhh2, 2021-08-11:
I try another map called Shalun , it also has the same problem ,so do you know how to fix it?
[image]

EricBoiseLGSVL, 2021-09-03:
We are looking into this issue. Sorry about the late response.