lgsvl/simulator #1676: map is worng in apollo's dreamview

OYhhh2, 2021-08-10:
the instructions to add other maps in svl to the apollo (my verision is apollo 6.0)
[image]

However,the map seems wrong ,the car is not in the right possition in dreamview , the simulator has no problem, It seems that the map have a rotation ? i don't know what' s the problem. the map is SanFrancisco,and Shalun also havs this problem
[image]

Can you help me ?Thanks!

jasonliu00, 2021-09-13:
@OYhhh2  I have the same problem with you

EricBoiseLGSVL, 2021-09-13:
We are looking into this, thanks for posting

EricBoiseLGSVL, 2021-09-15:
This has been fixed and will be in our next release.  Thanks for posting

closed 2021-09-15 by EricBoiseLGSVL