lgsvl/simulator #1677: Sim starts sometimes in non-deterministic way when NPC-s involved

krtt, 2021-08-11:
I have an API Only sim and a py script. When there are no NPC-s in the sim, the sim starts always the same way and the car starts driving when expected. But when I add a NPC, for example one pedestrian, the sim sometimes starts differently. The difference can be that sometimes the car moves before it actually starts driving and goes for example off the lane so it will not start driving at all. Not sure if NPC-s have anything to do with it but it sometimes just looks this way. We need the sim to always start the same way. How to ensure that?

EricBoiseLGSVL, 2021-08-11:
We have a bug for this and we are working on it.  Turns out waypoint following for pedestrians are using NavAgent class to pathfind.  We hope to have it fixed by next release.

krtt, 2021-08-13:
I tried and at least 10 times in a row with NPC vehicle "Sedan" I don't see this. 

No, I am not supplying a seed. But I read about this option and I tried it and there was no difference with NPC sedan vehicle. It seems to be deterministic with the vehicles. At least the start of the sim and spawning the agents.

When do you plan the next release? For the pedestrians to be deterministic too.

EricBoiseLGSVL, 2021-08-16:
We plan for it to release with 2021.3, coming soon.  It requires a bit of a refactor unfortunately.

krtt, 2021-09-20:
Now I can unfortunately confirm that also without pedestrians the sim sometimes starts in non-deterministic way. I also tried supplying a seed, but the result is the same. For example the ego car just rolls on the street until Apollo starts controlling it. How to avoid this? I need the ego to stand still until Apollo controls it to drive.

EricBoiseLGSVL, 2021-09-20:
Yes, we do not make the ego deterministic.  It is in fact impossible unless Apollo or Autoware support it.  I recommend adding parking brake functionality until the AD stack is ready.

krtt, 2021-09-25:
Can you suggest an example of a good implemented parking brake functionality? Or where to start? Thanks.

EricBoiseLGSVL, 2021-09-28:
You could create/modify a sensor to apply the hand brake on start and disable the brake once the first routing command is received.  Just look to a AD stack control sensor and parking brake functionality.