lgsvl/simulator #1682: Line problem when calling OpenDRIVE

honey5967, 2021-08-13:
Hi I am using SVL to load OpenDRIVE files.
If the current lane has more than one line mark, only the first line seems to be visible. Is there any way to solve this problem?

EricBoiseLGSVL, 2021-08-18:
Without more info it is very difficult to solve this since open drive maps can vary so much for each user.  Are you just loading OpenDrive maps?  What are you trying to do?
What do you mean "one lane mark"? Do you mean double lines?  If so, we don't support that in the mesh generation tool as of yet?  Is this what you mean?

We have been working on improving OpenDrive support and have a few fixes in our 2021.3 release. Hopefully this will improve support.

honey5967, 2021-08-19:
> 오픈 드라이브 맵은 사용자마다 매우 다양하기 때문에 추가 정보가 없으면 이 문제를 해결하기가 매우 어렵습니다. OpenDrive 지도를 로드하고 있습니까? 당신은 무엇을 하려고 합니까?
> "1차선 표시"은(는) 무슨 뜻인가요? 이중선을 말씀하시는건가요? 그렇다면 아직 메쉬 생성 도구에서 지원하지 않습니까? 이것이 네가 말하는 뜻 이니?
> 
> 우리는 OpenDrive 지원을 개선하기 위해 노력하고 있으며 2021.3 릴리스에서 몇 가지 수정 사항이 있습니다. 이것이 지원을 개선할 수 있기를 바랍니다.

> Without more info it is very difficult to solve this since open drive maps can vary so much for each user. Are you just loading OpenDrive maps? What are you trying to do?
> What do you mean "one lane [...]

Yes I want to visualize the road by loading an OpenDRIVE file from SVL.
However, when two or more roadMarks come in from the lane, all the roadMark images are not applied, so I inquired because I was wondering if this would be extended.
Take the code as an example
                     <lane id="0" type="none" level="false">
                         <roadMark sOffset="0.00000000000000000e+0" type="solid solid" material="standard" color="yellow" width="1.25000000000000000e-1" laneChange="none"/>
                         <roadMark sOffset="2.8500000000000004e+1" type="none" material="standard" color="white" laneChange="none"/>
                         <userData/>
                     </lane>
If this code comes in, it seems that the image of all lines is applied based on the first roadMark information, and the second roadMark information is missing.
Maybe it's something I'm not familiar with?
Thankyou!!

honey5967, 2021-08-19:
> Without more info it is very difficult to solve this since open drive maps can vary so much for each user. Are you just loading OpenDrive maps? What are you trying to do?
> What do you mean "one lane [...]

I applied the image rotation today and if it works as intended, it seems that the crosswalk can be created normally. As for the lane change problem, can you please solve the crosswalk problem and then talk again?
Yes I want to visualize the road by loading an OpenDRIVE file from SVL.
However, when two or more roadMarks come in from the lane, all the roadMark images are not applied, so I inquired about whether to expand this.
Take the code as an example

[image]

If this code comes in, it seems that the image of all lines is applied based on the first roadMark information, and the second roadMark information is missing.
Maybe it's something I'm not familiar with?
Thankyou!!

EricBoiseLGSVL, 2021-08-19:
Thanks for more info.  This helps immensely and glad to help.  You are helping others by making better features. :)
What do you mean solve the crosswalk? You say that it can be created normally?
Can you please post an image of the imported mesh so we can see from the code?
I'll have our team look at this but to fix the issues, please be very clear and use images to show issues for visual problems.

honey5967, 2021-08-19:
> 더 많은 정보를 감사합니다. 이에 대해 기쁘게 생각합니다. 더 나은 기능을 사용하세요. :)
> 많은 사람들에게 도움이 되었습니까? 이 정도는?
> 그와 관련하여 이미지를 게시할 수 있습니까?
> 이 문제를 보다 중요하게 생각하십시오.

Oh, I'm sorry, Eric. It wasn't about crosswalks. my mistake.
What I want to ask is that of the two roadMarks, only the first roadMark applies to the lane specified in the code above. Do you have any plans to expand to apply sOffset to generate all roadMark information for the lane?

sorry for the confusion

EricBoiseLGSVL, 2021-08-19:
No worries, we will look into this, thanks for posting