lgsvl/simulator #1683: Ego Vehicle Build and Upload

ertauren, 2021-08-13:
Hi,
I build and used my ego vehicle asset with version 2021.1. I decided to upgrade it to 2021.2 and built with it. It didn't give any errors at unity. But when i tried to upload it to website it gave "assetUploader.errors.(undefined)" once upload finished and failed.

Did anyone else had this problem? Anyone have a suggestion i can try? Is there limit of car assets my account can have?

EricBoiseLGSVL, 2021-08-18:
Yes, we have this issue and should have fixes releasing soon. Sorry about the confusion.  When in doubt, delete the old vehicle and upload.  Then if prompted, make a new vehicle.

EricBoiseLGSVL, 2021-09-02:
Please see #1706

the current maximum for the GLB size (inside the asset bundle) is set to 50 MB. But we are increasing it to 200 MB in the upcoming release (that is going out this month end).
The appropriate error message will also be available in the upcoming release.

closed 2021-09-02 by EricBoiseLGSVL