lgsvl/simulator #1684: Cannot use bridges on custom development versions?

dheerajakula, 2021-08-13:
Hi I am unable to use bridges on development version, because development is not a compatible simulator version.

[image]

dheerajakula, 2021-08-18:
I found how to resolve the issue we can take the bridge plugins [https://www.svlsimulator.com/docs/plugins/bridge-plugins/](url) route and use the opensource code for example the code for cyber rt is at [https://github.com/lgsvl/CyberRT](url). If using editor just place this code at Assets/External/Bridges/ to use this bridge don't need to build explicitly.

It would be great if SVL supports bridges on development versions instead of building these plugins.

heeen, 2021-08-18:
We have an upcoming feature to send a version other than Development from editor and also bridges should support Development soon I believe

EricBoiseLGSVL, 2021-08-18:
Yep, coming next release.