lgsvl/simulator #1685: Quickstart Scripts: PythonAPI

marcusvinicius178, 2021-08-13:
Hi at first I wish to congratulate the LG team (@EricBoiseLGSVL , @lemketron ) for the generation of the quickstart scripts on Python. It is outstanding and cover interesting scenarios!!!! 

I have runned all the scripts and had few issues with some of them. I hope to get some additional info here:

1) 
From Script 06: https://github.com/lgsvl/PythonAPI/blob/master/quickstart/06-save-camera-image.py

I wish to save the image in other path (not the standard path it is saving on svl simulation folder). I have then modified the script to this:

[image]

However the terminal raise this error:

`Python API Quickstart #6: Saving an image from the Main Camera sensor
Traceback (most recent call last):
  File "./06-save-camera-image.py", line 31, in <module>
    s.save("main-camera.png", autoware-auto-ros1/PythonAPI/quickstart/Apollo_Test_Cases
NameError: name 'autoware' is not defined
`

In addition I issue to know if this script take just one snapshot? Or I could do a for loop to take a lot of pictures maybe?

Still about sensor data and script:

2) The 07 script (PCD data): https://github.com/lgsvl/PythonAPI/blob/master/quickstart/07-save-lidar-point-cloud.py

I tried to execute it but I get this error:

[image]

I think there is an error with the code implemented...**is it deprecated** with the new SVL releases? Also the same question as the previous regarding the data,. The script takes just one snapshot? Must I do a for loop or another program to take continuous data? **I wish to generate a video from PCD, is it possible? I know that for the image I have the video recorder sensor...but for lidar?**

3- The script related to follow waypoints (13, 25 and 26). I have a programming doubt:

It is commented in their lines: "# When the NPC is within 0.5m of the waypoint, this will be called" on script 13
and  "# When the NPC is within 0.5m of the waypoint, this will be called" on scripts 25 and 26

Actually I have read the code and did not get the idea and couldn't see the math that enables the callback to  be called 0.5 meters before reach the waypoint and for the others scripts 1 meter before reach the waypoint. Where is this specifically specified in the block of codes? I would like to understand the math behind...

4- on this script: https://github.com/lgsvl/PythonAPI/blob/master/quickstart/98-npc-behaviour.py

I would like to add the **DrunkerDriver and the Trailer **  hahahaha to play around a little bit. Where can I get these characters? I didn't find these vehicle on simulator webpage to add...

And one last question not directly related to Python Scripts. But to Localization on AutowareAI with SanFrancisco LGSVL map. On ROS visualizer (Rviz) it is necessary to tell the robot where it is located using the **2D EstimatePose** arraw icon.
However the San Franscisco map is is unbelievably huge and the streets pretty similar to define where this robot is...

[image]

I could not set the arrow to the correct spot, then I cannot keep working with the simulation....
I ask you guys if **is there a method to localize this car on rviz2 (autoware). Is there a tool to convert the LGSVL coordinates to the Rviz2 coordinates. I even does not know which coordinates Rviz uses....probably NOT GPS** If you be aware of some strategy please let me know. Otherwise the SanFrancisco map is useless to be used with Autoware.

Thanks in advance!

heeen, 2021-08-16:
1: you need to add quotes around the path string like so "/home/autoware-auto-ros1..."

4: Drunk driver source is here: https://github.com/lgsvl/BehavioursNPC trailer model does not seem to be public right now.

EricBoiseLGSVL, 2021-08-16:
2: This may be a bug.  @kamkolak-lge can you please confirm this we can roll this into the point cloud support
3: These checks are done in the waypoint following behavior. It is just a simple distance check
5: We don't have this functionality.  We expect the user to stop the simulation to close connections. @lishammel This could be brought up in planning if it fits?
Last one: Not sure on this.  We are working on a NavOrigin feature to localize for Nav2 but I'd need to ask @hadiTab or @cyberphysicalrobotics for help localizing with RViz2.

ToniHille, 2021-09-13:
Hello, any Update on Question 2.? 
I have the same error message when trying to save lidar files using the newest release.

EricBoiseLGSVL, 2021-09-15:
We are working on fixing issue 2.  Looks like our refactor of Lidar broke this quick script.  In the meantime we suggest using the bridge to data collect.  The api is not efficient to do this, especially with large data sets from lidar.

kamkolak-lge, 2021-09-15:
Regarding question 2. - it was confirmed to be a bug and the fix is already pending review. It should be available soon.

EricBoiseLGSVL, 2021-09-16:
Closing this because the multiple questions have been answered.  Let's try to open one issue per issue.  These multiple issue ones are difficult to track and parse, thanks

closed 2021-09-16 by EricBoiseLGSVL